micropython/micropython #11318: esp32c3 BLE gatts_notify can't send data OSError: -128

QGB, 2023-04-23:
```
from machine import Pin, Timer
from time import sleep_ms
import ubluetooth

class BLE():
	def __init__(self, name):   
		self.name = name
		self.ble = ubluetooth.BLE()
		self.ble.active(True)

		self.led = Pin(12, Pin.OUT)
		self.timer1 = Timer(0)
		
		self.disconnected()
		self.ble.irq(self.ble_irq)
		self.register()
		self.advertiser()

	def connected(self):		
		self.timer1.deinit()
	
	def blink(self,a):
		'''a= Timer(3fca88b0; alarm_en=1, auto_reload=1, counter_en=1) '''
		# print(a)
		self.led(1)
		sleep_ms(200)
		self.led(0)
	def disconnected(self):		
		self.timer1.init(period=1000, mode=Timer.PERIODIC, callback=self.blink )

	def ble_irq(self, event,data):
		print(event,data)
		if event == 1:
			'''Central disconnected'''
			self.connected()
			self.led(1)
		
		elif event == 2:
			'''Central disconnected'''
			self.advertiser()
			self.disconnected()
		
		elif event == 3:
			'''New message received'''			
			buffer = self.ble.gatts_read(self.rx)
			message = buffer.decode('UTF-8').strip()
			print(message)

    def register(self):		
		# Nordic UART Service (NUS)
		NUS_UUID ='6E400001-B5A3-F393-E0A9-E50E24DCCA9E'
		RX_UUID = '6E400002-B5A3-F393-E0A9-E50E24DCCA9E'
		TX_UUID = '6E400003-B5A3-F393-E0A9-E50E24DCCA9E'
			
		BLE_NUS = ubluetooth.UUID(NUS_UUID)
		BLE_RX = (ubluetooth.UUID(RX_UUID), ubluetooth.FLAG_WRITE)
		BLE_TX = (ubluetooth.UUID(TX_UUID), ubluetooth.FLAG_NOTIFY)
			
		BLE_UART = (BLE_NUS, (BLE_TX, BLE_RX,))
		SERVICES = (BLE_UART, )
		((self.tx, self.rx,), ) = self.ble.gatts_register_services(SERVICES)

	def send(self, data):
		self.ble.gatts_notify(0, self.tx, data + '\n')#OSError: -128

```
test
```
import ble

>>> self=ble.ble
>>> self.ble.gatts_notify(0, self.tx, b'2')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -128
>>>
```

compile from latest git repo

MicroPython 9e6885a-dirty on 2023-04-23; ESP32C3 module with ESP32C3

QGB, 2023-04-23:
https://github.com/orgs/micropython/discussions/10736

hamedf62, 2023-05-04:
you have to specify conn_handler in gatts_notify() method, which is not 0 for mine, but don't know how to get this conn id dynamically per connection?

QGB, 2023-05-04:
Is it possible to gatts_notify all client ?

towelbro0812, 2023-05-10:
You can try to change 
> self.ble.gatts_notify(0, self.tx, b'2')
to 
> self.ble.gatts_notify(1, self.tx, b'2')

peterschlem, 2023-06-07:
A life saver changing the channel to 1.  I am guessing there may be two channels 0 for rcv and 1 for tx.

ssddero1, 2023-06-20:
esp32 s3  too ,if conn_handler =0,then #OSError: -128,if conn_handler =1,then send nothing(not Error,not send ).

ssddero1, 2023-06-20:
> esp32 s3 too ,if conn_handler =0,then #OSError: -128,if conn_handler =1,then send nothing(not Error,not send ).

sorry,My fault，if conn_handler =1，it is ok。

MATTYGILO, 2023-09-05:
Make sure that the device you are sending data to has not disconnected from the ESP32. It preferable for the ESP32 to sever any BLE connections

magixyu, 2023-12-25:
> You can try to change
> 
> ```
> self.ble.gatts_notify(0, self.tx, b'2')
> ```
> 
> to
> 
> ```
> self.ble.gatts_notify(1, self.tx, b'2')
> ```
> 
> I used this method to solve this problem

In "ble_irq" unpack the data in the related event, like "event == 3:"
`
conn_handler, attr_handle = data
`
Then use the conn_handler in gatts_notify
`
gatts_notify(conn_handler, self.tx, data)
`

Tristan

zoland, 2024-03-23:
I have the same problem on MicroPython v1.22.0 on 2023-12-27; Generic ESP32 module with ESP32, conn_handler=0, but gatts_notify(1, self.tx, data) does not change the result: OSError -128

zoland, 2024-03-23:
Solved. First - Write data, then Notify WITHOUT data.
By the way can be checked by is_connected() before

        self.ble.gatts_write(self.tx, data + '\n')
        
        if self.conn_handle is not None:
            self.ble.gatts_notify(self.conn_handle, self.tx)

qiujiejia, 2024-09-28:
● below is my code , it has been tested in WeMos LOLIN32 Lite  and esp32-c3 mini board . 

● for esp32-c3 mini  , send message ,should use :
 self.ble.gatts_notify( self.conn_handle , self.tx, data.encode('UTF-8') )  
not use 
self.ble.gatts_notify(0, self.tx, data.encode('UTF-8') )

``` 

'''
can only receive 20 char or 6 chinese char   

Explanation of the Limitations:

    MTU Size: The default MTU size for BLE on many devices, including the ESP32, is often 23 bytes, but only 20 bytes can be used for actual payload data. The remaining bytes are used for overhead (such as attribute handles).

Character Encoding:

    ASCII Characters: Each ASCII character takes up 1 byte. Therefore, you can send a maximum of 20 ASCII characters in one BLE packet.
    Chinese Characters: These are typically encoded in UTF-8 and may take up more than 1 byte (often 3 bytes or more). Consequently, if you send more than 6 Chinese characters, the total byte count exceeds the MTU limit, leading to errors.

'''

from machine import Pin
from machine import Timer
import time
# import ubluetooth as bluetooth
import bluetooth

class ESP32_BLE():
    def __init__(self, name):

        self.conn_handle = None  # Add this to track the connection handle
        
        self.ble = bluetooth.BLE()  # 创建BLE对象

        self.ble.config(gap_name=name)  # 配置蓝牙，给蓝牙起个名字
        self.name = name  # 传进来的参数name

        self.ble.irq(self.ble_irq)  # 蓝牙调用中断函数

    def start(self):
        self.ble.active(True)  # 启动蓝牙  
        self.register()  # 注册服务
        self.advertiser()  # 开始广播

    def register(self):

        service_uuid = bluetooth.UUID('6E400001-B5A3-F393-E0A9-E50E24DCCA9E')  # Generic Access Service
        tx_uuid = bluetooth.UUID('6E400002-B5A3-F393-E0A9-E50E24DCCA9E')  # Appearance Characteristic
        rx_uuid = bluetooth.UUID('6E400003-B5A3-F393-E0A9-E50E24DCCA9E')  # Device Name Characteristic

        services = (
            (service_uuid, (
                (tx_uuid, bluetooth.FLAG_NOTIFY),  # 发送通知
                (rx_uuid, bluetooth.FLAG_WRITE),   # 接收写入
            )),
        )

        ((self.tx, self.rx), ) = self.ble.gatts_register_services(services)

    def advertiser(self):  # 当断开连接时，开始广播
        name = bytes(self.name, 'UTF-8')  # 将设备名字转为utf-8格式

        adv_data = bytearray()
        adv_data += bytearray([0x02, 0x01, 0x06])  # 标准 BLE 广播数据头部
        adv_data += bytearray([len(name) + 1, 0x09]) + name  # 完整名称字段（0x09 = Complete Local Name）

        self.ble.gap_advertise(500, adv_data)  # 广播间隔  毫秒
        print("Advertising data:", adv_data)

    def send(self, data): 
        if self.tx is not None and self.conn_handle is not None :
            # 这两行代码的主要区别在于self.conn_handle的使用。第一行中的0通常表示向所有连接的设备发送通知，而第二行中的self.conn_handle是指定的连接句柄，意味着通知将仅发送给特定的已连接设备。这样，第二行更具针对性。
          # self.ble.gatts_notify(0, self.tx, data.encode('UTF-8') )  # 蓝牙发送通知信息
            self.ble.gatts_notify( self.conn_handle , self.tx, data.encode('UTF-8') )  # 蓝牙发送通知信息  esp32-c3 mini 必须使用 self.conn_handle

    def send_large_message(self, message):
        max_chunk_size = 30  # Reduced chunk size to leave room for BLE overhead
        for i in range(0, len(message), max_chunk_size):
            chunk = message[i:i + max_chunk_size]
            try:
                self.send(chunk)  # Sending the chunk
                time.sleep(0.010)        # 必须延迟一下,不然连续发送buffer不够会出错 (停1ms还是会出错)
            except OSError as e:
                print(f"Failed to send chunk due to {e}")
                break  # Exit if there's an error

    def ble_irq(self, event, data):
        
        # print("BLE IRQ Event:", event, "Data:", data)  # 打印事件及数据以帮助调试
        
        if event == 1:  # 手机已连接此设备
            print("Connected")
            self.conn_handle, _, _ = data  # Store the connection handle
        elif event == 2:  # 手机已断开此设备
            print("Disconnected")
            self.conn_handle = None  # Reset connection handle
            self.advertiser()  # 重新广播
        elif event == 3:  # 手机发送数据给此设备
            if self.rx is not None:  # 确保 self.rx 已初始化
                buffer = self.ble.gatts_read(self.rx)  # 接收来自手机的数据  can only receive 20 char or 6 chinese char  ， 查看文件顶部的说明   
                BLE_MSG = buffer.decode('UTF-8')  # 解码数据
                self.ReceiveMsg(BLE_MSG)
                # print("Received message:", BLE_MSG)

    # 虚函数
    def ReceiveMsg(self,data):
        pass

    def IsConnected(self):
        return self.conn_handle != None

    # Method to disconnect actively
    def disconnect(self):
        if self.conn_handle is not None : # ble_irq will set self.conn_handle = None
            print("Disconnect")
            self.ble.gap_disconnect(self.conn_handle)  # Actively disconnect the connection

    def stop(self): 
        self.disconnect()
        time.sleep(3)       #must deleay
        self.ble.active(False)  # 关闭蓝牙
        print("Bluetooth has been turned off to reduce power consumption.")

#━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ the test code 
class MyBLe(ESP32_BLE):
    def ReceiveMsg(self,data):
        print("Received message:", data)
        self.send(data)
        # ble.send_large_message  这里直接调用 ble.send_large_message 可能会内存出问题
        # self.disconnect()

ble = MyBLe("ESP32 BLE")  # 创建一个对象，带1个实参，后面用来给BLE起名字用
ble.start()

while True:

    '''
    if ble.IsConnected():
        ble.send("is connected")
    '''

    time.sleep(2)
#━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━

 ```

projectgus, 2025-08-26:
Looks like the original author has resolved the issue, and the other comments are all solved via a similar method - so I'm going to close this issue.

closed 2025-08-26 by projectgus