micropython/micropython #11337: RP2 Viper issues

BrianPugh, 2023-04-26:
Firmware: `MicroPython v1.19.1-1019-g9e6885ad8 on 2023-04-24; Raspberry Pi Pico with RP2040`

Not really sure how to summarize what precisely the issue is, but here's my minimal working example:

```
from io import BytesIO

@micropython.viper
def foo(f):
    full_mask = int(0xFFFFFFFF)

    bit_buffer = 0
    bit_pos = 0

    val = 0

    for i in range(10):
        print(bin(i))  # Just to demonstrate this works fine

    print("this stuff is broken:")
    while True:
        if bit_pos == 0:
            try:
                bit_buffer |= int(f.read(1)[0]) << 24
                bit_pos += 8
            except IndexError:
                break
        val = (val << 1) | (bit_buffer >> 31)
        bit_buffer = (bit_buffer << 1) & full_mask
        bit_pos -= 1
        print(bin(val))

with BytesIO(b"\x55") as f:
    foo(f)
```

Output; I would expect the printed output to be the shifted in bits of `0x55`
```
0b0
0b1
0b10
0b11
0b100
0b101
0b110
0b111
0b1000
0b1001
this stuff is broken:
0b0
-0b1
-0b10
-0b1
-0b10
-0b1
-0b10
-0b1
```

Related context: I'm doing a lot of bit-buffering, and something is incorrect here.

peterhinch, 2023-04-26:
I've not studied this in detail but the line
```python
full_mask = int(0xFFFFFFFF)
```
rings alarm bells. Please see https://github.com/micropython/micropython/issues/9114

BrianPugh, 2023-04-26:
thank you so much @peterhinch! I believe that's causing my issues. We can close this issue if we think it's sufficiently captured by existing issue #9114

robert-hh, 2023-04-26:
Since you opened the issue, you can close it as well.

closed 2023-04-26 by BrianPugh