micropython/micropython #11340: website: update footer, contact, links etc

andypiper, 2023-04-26:
I noticed a few issues around the main micropython.org website that need attention to align with the current state of the project.

Looking at the footer:

- MicroPython column on the left is good ✅ 
- Resources column
  - https://micropython.org/help/ is hugely out of date. Is that page needed now?
  - suggest relabelling Tutorials as "Pyboard" as it links to Pyboard-specific information
- Contact column
  - Contact us page https://micropython.org/contact/ still links to the old archived forum and to the IRC channel (both of these should be updated to new Discussions and Discord). Add Mastodon.
  - IRC link might be redundant, is it still used? There's a Discord link in the header and in the left MicroPython column.

As an observation, the site serves a couple of purposes which I think drives some of these inconsistencies - specifically, a home for the MicroPython language, and a home / store for the PyBoard, so for example the tutorials etc (and the Help! page) are aimed at the PyBoard.

Gadgetoid, 2023-04-27:
Perhaps a controversial suggestion, but would it make sense to migrate the site to a public `micropython-org` GitHub repository so the community can take a more active role in maintaining and correcting it? Right now some of this seems to have leaked out onto the Wiki which is a little chaotic.

A lot of the store items seem to be out of stock. Is it winding down? It still has a "COVID-19" message and the store top-nav differs from the main site top nav.

jonnor, 2024-08-24:
The help page now redirects to docs, that is good.

However, the contact page still links to the old forum.
And the https://micropython.org/live/ linked on the frontpage has been down for years.

How can the community help out with the website?  @dpgeorge @jimmo

andypiper, 2024-08-24:
Not sure whether @mattytrentini can assist on this as well? It's been a while since this issue was opened.