micropython/micropython #11346: ussl_basic.py.exp test results are wrong for calling close twice?

dlech, 2023-04-26:
In `tests/extmod/ussl_basic.py`, we call `close()` twice.

https://github.com/micropython/micropython/blob/294baf52b346e400e2255c6c1e82af5b978b18f7/tests/extmod/ussl_basic.py#L54-L57

This prints `TestSocket.ioctl 4 0` because of:

https://github.com/micropython/micropython/blob/294baf52b346e400e2255c6c1e82af5b978b18f7/tests/extmod/ussl_basic.py#L18-L20

However, in the .exp file, we are only expecting it to print once (and CI is passing with this):

https://github.com/micropython/micropython/blob/294baf52b346e400e2255c6c1e82af5b978b18f7/tests/extmod/ussl_basic.py.exp#L5

Using one of our Pybricks builds, we are getting the output printed twice though, which seems expected since `close()` was called twice.

```
--- /home/runner/work/pybricks-micropython/pybricks-micropython/micropython/tests/results/extmod_ussl_basic.py.exp	2023-04-26 22:57:31.882920388 +0000
+++ /home/runner/work/pybricks-micropython/pybricks-micropython/micropython/tests/results/extmod_ussl_basic.py.out	2023-04-26 22:57:31.882920388 +0000
@@ -3,5 +3,6 @@
 TestSocket.setblocking(False)
 TestSocket.setblocking(True)
 TestSocket.ioctl 4 0
+TestSocket.ioctl 4 0
 OSError: read
 OSError: write
```

What is the difference? Our build uses axtls instead of mbed, but since both implementations use the standard MicroPython stream implementations for close, etc. I don't see what the difference could be.

dlech, 2023-04-26:
Related change: https://github.com/micropython/micropython/pull/8916/commits/18ecc29bb8eedcc1f4fd999310467f0942eeaaac

dlech, 2023-04-26:
Ah, I think we need to make this matching change to axtls: https://github.com/micropython/micropython/commit/ed58d6e4ce5878d5f53457e72686962e7f57f5df

dpgeorge, 2023-04-27:
This test depends a lot on the underlying implementation.  We can change the test to be more flexible and pass on both axtls and mbedtls.  But would be good to first understand why exactly the difference is there.

dlech, 2023-04-27:
I think the change was caused by this line:

https://github.com/micropython/micropython/blob/867e4dd3dc9235974974fd5dab204371616d6f49/extmod/modussl_mbedtls.c#L395

After the first call to close, the wrapped socket object is released so the second call to close doesn't get passed to the wrapped socket since there is no longer a reference to it.