micropython/micropython #11352: Adjust help() command feedback for STM32 ports

Yannick-Marietti, 2023-04-27:
Hello !

Not really a bug (technically), but a report about wrong `help()` command output when it is run on and STM32 board (except Pyboard of course). I tested it on NUCLEO-WB55 firmware.

This is `help()` output on  NUCLEO-WB55 firmware:

```
>>> help()
Welcome to MicroPython!

For online help please visit http://micropython.org/help/.

Quick overview of commands for the board:
  pyb.info()    -- print some general information 
  pyb.delay(n)  -- wait for n milliseconds
  pyb.millis()  -- get number of milliseconds since hard reset
  pyb.Switch()  -- create a switch object
                   Switch methods: (), callback(f)
  pyb.LED(n)    -- create an LED object for LED n (n=1,2,3,4)
                   LED methods: on(), off(), toggle(), intensity(<n>)
  pyb.Pin(pin)  -- get a pin, eg pyb.Pin('X1')
  pyb.Pin(pin, m, [p]) -- get a pin and configure it for IO mode m, pull mode p
                   Pin methods: init(..), value([v]), high(), low()
  pyb.ExtInt(pin, m, p, callback) -- create an external interrupt object
  pyb.ADC(pin)  -- make an analog object from a pin
                   ADC methods: read(), read_timed(buf, freq)
  pyb.DAC(port) -- make a DAC object
                   DAC methods: triangle(freq), write(n), write_timed(buf, freq)
  pyb.RTC()     -- make an RTC object; methods: datetime([val])
  pyb.rng()     -- get a 30-bit hardware random number
  pyb.Servo(n)  -- create Servo object for servo n (n=1,2,3,4)
                   Servo methods: calibration(..), angle([x, [t]]), speed([x, [t]])
  pyb.Accel()   -- create an Accelerometer object
                   Accelerometer methods: x(), y(), z(), tilt(), filtered_xyz()
```

Commands 

- pyb.info()
- pyb.delay(n)
- pyb.millis()
- pyb.Switch()
- pyb.Servo(n)
- pyb.Accel()

do not seem to be available.
Is it possible to remove them from   `help()` feedback ?

Also, DAC is not available on all STM32 boards, so I would rather write:

```
pyb.DAC(port) -- make a DAC object (if supported by your board MCU)
                   DAC methods: triangle(freq), write(n), write_timed(buf, freq)
```

Best regards

Yannick

mattytrentini, 2023-05-03:
Yes, you're correct. 

For reference, the help message is a [static string for the stm32 port](https://github.com/micropython/micropython/blob/master/ports/stm32/help.c#L29). This made sense when `pyb` was _always_ present for the stm32 port but that changed when the `MICROPY_PY_PYB_LEGACY` define was introduced - and indeed for the WB55 `pyb` [is disabled](https://github.com/micropython/micropython/blob/master/ports/stm32/boards/NUCLEO_WB55/mpconfigboard.h#L8).

We ought to check the state of `MICROPY_PY_PYB_LEGACY` in [help.c](https://github.com/micropython/micropython/blob/master/ports/stm32/help.c) and use it to determine the contents of the help message.

dpgeorge, 2023-06-02:
Thanks for the report.

Should be fixed by aac8744671b690e22088a2960339b43c0bd45d88 and e4886dda859d873e77b1ff5f5b8d4a3aa0f3d517

closed 2023-06-02 by dpgeorge