micropython/micropython #11354: REPL / built-in readline: Add CTRL-L support (clear screen)

neuschaefer, 2023-04-27:
Right now, the built-in readline implementation ([`shared/readline/readline.c`](https://github.com/micropython/micropython/blob/master/shared/readline/readline.c)) does not support CTRL-L, which is a somewhat common key to clear the the screen, and move the prompt and command line to the top of the screen. At least [bash supports it](https://ss64.com/bash/syntax-keyboard.html).

This feature could be optional, and features like `MICROPY_REPL_EMACS_KEYS` already are, but the size savings wouldn't very much.

I might implement this myself If I get around to it. If anyone else want to do it, please let me know here.

dpgeorge, 2023-04-27:
This feature is actually implemented but in a library: `upysh.clear` (installable via `mpremote mip install upysh`, for example).  See https://github.com/micropython/micropython-lib/blob/master/micropython/upysh/upysh.py

neuschaefer, 2023-04-27:
> This feature is actually implemented but in a library: `upysh.clear` (installable via `mpremote mip install upysh`, for example). See https://github.com/micropython/micropython-lib/blob/master/micropython/upysh/upysh.py

As far as I can see, the upysh library doesn't hook into the readline implementation at all. My aim here is not to clear the screen from python code, not even to type `clear()` on the command line, but to replicate bash's convenience feature of being able to quickly clear the screen and move the current line of input to the top. Anything that isn't hooked into the readline implementation can't quite achieve that.

So, `upysh` is at best a reference for the control sequence that needs to be sent to the terminal.