micropython/micropython #11355: The mbedtls submodule targets an old, unsupported branch

gilles-peskine-arm, 2023-04-27:
Micropython currently pulls in Mbed TLS at commit https://github.com/ARMmbed/mbedtls/tree/1bc2c9cb8b8fe4659bd94b8ebba5a4c02029b7fa which is from the Mbed TLS 2.16 branch. Support for Mbed TLS 2.16 ended in January 2022. An old branch of a security library is very likely to have known vulnerabilities.

Please upgrade to a [maintained version of Mbed TLS](https://github.com/Mbed-TLS/mbedtls/blob/development/BRANCHES.md). At the moment, the choices are:

* `mbedtls-2.28` — long-time support branch, only getting bug fixes. We plan to support this branch until December 2024. APIs are fully compatible with 2.16.
* `development` — development branch, with new features (which can mean an increase in code size). This is a new major version of Mbed TLS (3.x series) compared to the current one, so some application code may need to be rewritten due to API changes.

Furthermore, in the future, to avoid shipping software that embeds a library version which may have known vulnerabilities, please upgrade Mbed TLS regularly. Releases ship [through GitHub](https://github.com/Mbed-TLS/mbedtls/releases) and are announced on the [mbed-tls-announce mailing list](https://lists.trustedfirmware.org/mailman3/lists/mbed-tls-announce.lists.trustedfirmware.org/).

dpgeorge, 2023-04-27:
Thanks for the information.  There was a start to update to Mbed TLS 3.2.1 in #8988, and now that v1.20.0 is released we can make progress with that PR and follow through with the update.

Carglglz, 2023-04-27:
@dpgeorge in the meantime an to update to `mbedtls-2.28` wouldn't be a bad idea, it was pretty straightforward i.e.
just a git check-out and including/removing some components in `extmod.*` make files.
,.
```diff
diff --git a/extmod/extmod.cmake b/extmod/extmod.cmake
index 7cf16fa1b..b053b3aef 100644
--- a/extmod/extmod.cmake
+++ b/extmod/extmod.cmake
@@ -154,7 +154,7 @@ if(MICROPY_SSL_MBEDTLS)
         ${MICROPY_LIB_MBEDTLS_DIR}/library/md4.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/md5.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/md.c
-        ${MICROPY_LIB_MBEDTLS_DIR}/library/md_wrap.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/oid.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/padlock.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/pem.c
@@ -179,9 +179,11 @@ if(MICROPY_SSL_MBEDTLS)
         ${MICROPY_LIB_MBEDTLS_DIR}/library/ssl_cli.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/ssl_cookie.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/ssl_srv.c
+        ${MICROPY_LIB_MBEDTLS_DIR}/library/ssl_msg.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/ssl_ticket.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/ssl_tls.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/timing.c
+        ${MICROPY_LIB_MBEDTLS_DIR}/library/constant_time.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/x509.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/x509_create.c
         ${MICROPY_LIB_MBEDTLS_DIR}/library/x509_crl.c
diff --git a/extmod/extmod.mk b/extmod/extmod.mk
index e895c8169..259ceea91 100644
--- a/extmod/extmod.mk
+++ b/extmod/extmod.mk
@@ -172,7 +172,6 @@ SRC_THIRDPARTY_C += $(addprefix $(MBEDTLS_DIR)/library/,\
        md4.c \
        md5.c \
        md.c \
-       md_wrap.c \
        oid.c \
        padlock.c \
        pem.c \
@@ -197,9 +196,11 @@ SRC_THIRDPARTY_C += $(addprefix $(MBEDTLS_DIR)/library/,\
        ssl_cli.c \
        ssl_cookie.c \
        ssl_srv.c \
+       ssl_msg.c \
        ssl_ticket.c \
        ssl_tls.c \
        timing.c \
+       constant_time.c \
        x509.c \
        x509_create.c \
        x509_crl.c \
```
All tests seem OK too. 👍🏼 (Tested with ports `unix` and `esp32` @ `esp-idfv4.4`)

dpgeorge, 2023-04-27:
@Carglglz OK I'm happy to do that, update to 2.28.  Do you want to submit a PR?  If not I'll get to it at some point.

Carglglz, 2023-04-28:
> @Carglglz OK I'm happy to do that, update to 2.28. Do you want to submit a PR? If not I'll get to it at some point.

Done!

dpgeorge, 2023-05-01:
mbedTLS was upgraded to v2.28.1 in 7e0a38f7f7eea2bf492fe641189c1da4bb8b30f1

gilles-peskine-arm, 2023-05-02:
Thanks, that's an improvement. But why 2.28.1? That's the long-time support _branch_, but not the latest release in that branch. The latest long-time support release, including security fixes, is currently [2.28.3](https://github.com/Mbed-TLS/mbedtls/releases/tag/v2.28.3).

dpgeorge, 2023-05-02:
> . The latest long-time support release, including security fixes, is currently [2.28.3](https://github.com/Mbed-TLS/mbedtls/releases/tag/v2.28.3).

OK, thanks for that info, I didn't realise.

@Carglglz could you please try and update it?

Carglglz, 2023-05-02:
@dpgeorge my bad, I didn't realise either there were newer `v.2.28.x` versions, luckily it is a simple tag check out this time 🤞🏼

dpgeorge, 2024-01-30:
MicroPython now uses mbedtls v3.5.1.

closed 2024-01-30 by dpgeorge