micropython/micropython #11357: Changing AP interface MAC address using bytes results in a crash, but bytearray works

cyberedgelord, 2023-04-27:
Hello!

We have a Python module named "test" and following code in `__init__.py`:
```
import network

network.WLAN(network.AP_IF).active(True)
network.WLAN(network.AP_IF).config(mac=bytearray(b'\x9e\x9c\x1f\x00\x00\x00'))  # <- Works
network.WLAN(network.AP_IF).config(mac=b'\x9e\x9c\x1f\x00\x00\x00')  # <- Crashes
```

After freezing "test" module into the firmware and running `import test`, the ESP8266/ESP8285 crashes, but when using bytearray works (thanks to @jimmo).

"Dirty" build of lovely MicroPython fork of @glenn20:
MicroPython v1.19.1-espnow-8-g8d6473316-dirty on 2023-04-27; ESP module (1M) with ESP8266

[test.zip](https://github.com/micropython/micropython/files/11348107/test.zip)

glenn20, 2023-04-28:
Similar behaviour occurs for the `esp_now_send()` function in the ESP8266 SDK which crash the device when passed a string stored in ROM (eg. when code is frozen or an interned string). In the ESPNow module, I detect when the data buffer of an object is not located in the heap, copy the data to a temporary buffer on the C stack, then pass that to `esp_now_send()`.

glenn20, 2023-04-28:
A simple way to trigger this error, without the need to freeze code is:
```python
import network
network.WLAN(network.AP_IF).config(mac='abcdef')    # Does not reboot - 'abcdef' is stored in heap.
network.WLAN(network.AP_IF).config(mac='config')    # Does reboot - 'config' is an interned string stored in ROM
```
I've tested this behaviour is replicated on micropython release images v1.19.1 and v1.20.0.