micropython/micropython #11367: esp8266: with micropython 1.20, mpy files aren't running

massimosala, 2023-04-29:
Official firwmare from https://micropython.org/download/esp8266/
MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266

pip install adafruit-ampy : ver 1.1.0

(I don't know if there is a official micropython tool to upload files)

```
>>> sys.implementation
(name='micropython', version=(1, 20, 0), _machine='ESP module with ESP8266', _mpy=9478)
>>> sys.implementation._mpy & 0xff
6

```

I call ampy as usual:
_ampy.exe -v -march=xtensa -o mpy/basic.mpy basic.py_

With mycropython 1.19.1, the mpy files run fine.
Now micropython bails out
_ValueError: incompatible .mpy file_

Questions:
1) Is it ok the microcode version is still 6 ?
Or the developers forgot to bump up the version?

2)  Is ampy up-to-date with micropython 1.20 ?

Best regards, Massimo

dpgeorge, 2023-04-29:
What is the contents of your `basic.py` file?

massimosala, 2023-04-29:
On the new esp2866 there isn't a basic.py
There is a boot.py

```
import esp
esp.osdebug(None)
#import uos, machine
#uos.dupterm(None, 1) # disable REPL on UART(0)
import gc
#import webrepl
#webrepl.start()
gc.collect()

```

I changed the file yesterday, trying to solve: uncommenting then line to disable osdebug. Useless.

With the previous firmware, the same software is working.

This is the main.py

`import main2
`

I use this method, because main2 is in mpy format and reading the docs I see that main.py cannot be compiled.

My project is divided in three files: main2.mpy, basic.mpy, urequest.mpy
main2 imports standard libraries (os, sys, i2c, ...) and the other two custom files.

Is there any conflict having my "basic.mpy" ?
Note: with the previous firmware, the same files run fine.

I can make tests or provide more information, let me know.
Thanks

dpgeorge, 2023-04-30:
I ask about `basic.py` because I want to know what is inside it, maybe it has some native code that requires a different version of `mpy-cross` to compile it.

For all of the `.mpy` files that you have on your esp8266, can you print out and report here the first 4 bytes of those files?

Really what would help here is if you could provide instructions and code so someone else can fully reproduce the problem.

massimosala, 2023-04-30:
Hi Damien

Thanks for the fast answer.

To be sure, today I take a brand new board, flash MicroPython v1.19.1 and upload my  mpy files: the program works.
I use the same tools (I mean the same verions of mpy-cross and ampy).

These are the header of my custom files:
main2.mpy
b'M\x06\x00\x1f'

basic.mpy
b'M\x06$\x1f'

urequest.mpy
b'M\x06\x00\x1f'

---

This is the file-system:

```
>>> os.listdir()
['assets', 'basic.mpy', 'boot.py', 'config.json', 'main.py', 'main2.mpy', 'urequest.mpy']

```

boot.py is a factory preset, I post its content in the previous comment.

main.py has only one line
`import main2
`

Have a nice Sunday!

dpgeorge, 2023-04-30:
> basic.mpy
> b'M\x06$\x1f'

This file has native xtensa code in it.  That's why I asked if you could provide the contents of the corresponding `basic.py` file.  The `basic.mpy` file needs to be regenerated with MicroPython v1.20.0 for it to work.

massimosala, 2023-04-30:
In basic.py for two function I use the decorator

`@micropython.native
`

I always generate the mpy files with
`mpy-cross-v6.exe -v -march=xtensa
`
before uploading.

1) I create againt the basic.mpy and upload it.
Not working.

2) Reading the docs, it seems the same MPY bytecode is compatible between versions.
If the version is 6 with 1.19 and 1.20 firmwares but the MPY files aren't compatible, I think it is misleading.
IMHO, breaking ABI / API changes have to increment the code version to 7.

3) Is mpy-cross-v6 up-to-date with the latest firmware ?
Pls try yourself with these configuration
```
main.py
        import main2
main2.mpy
        import basic.mpy
basic.mpy
       with any function using @micropython.native
```
and let me know.

massimosala, 2023-04-30:
Damien, you wrote:

>  The basic.mpy file needs to be regenerated with MicroPython v1.20.0 for it to work.

Is it possible to create the MPY files directly from micropython, without mpy-cross-v6.exe ?
If yes, is there micropython 1.20 for windows or linux ?
I cannot create the MPY files on the ESP, the automation tools run under a pc (at the moment Widows, but I can use also linux).

robert-hh, 2023-04-30:
You cannot create .mpy files using a micropython device. You have to use mpy-cross using Linux, OS X or Windows. If you have Linux, you can build mpy-cross yourself.

massimosala, 2023-04-30:
> You cannot create .mpy files using a micropython device. You have to use mpy-cross using Linux, OS X or Windows. If you have Linux, you can build mpy-cross yourself.

The last comment from Damien let me confused.
As I wrote, **I use mpy-cross-v6.exe**
Pls read this issue from the beginning, I am at a dead end with the last firmware.

robert-hh, 2023-04-30:
So I 
- made a simple python script witha` @micropthon.native` decorator called hello.py
- compiled mpy-cross and created a mpy file using `mpy-cross -march=xtensa hello.py`
- loaded the most recent firmware to a ESP8266 Wemos D1 Mini
- loaded the hello.mpy file to the device using mpremote and ran it.
No problem.
When you call mpy-cross --version, what do you get. Here the prompt is:
`MicroPython v1.20.0-25-gc2bf91219 on 2023-04-30; mpy-cross emitting mpy v6.1`

This is the hello.py script:
```
@micropython.native
def hello():
    print("Hello world, I'm a native")
    
hello()
```

massimosala, 2023-04-30:
Hi Robert

Thank for your support.

As I worte in my first comment, I install mpy-cross using pip install.

```
mpy-cross-v6.exe --version
MicroPython a327cfc on 2023-02-16; mpy-cross emitting mpy v6

```

As I assumed, there is a ABI change and the bytecode was changed from 6 to 6.1.

1) Where can I get the latest version of mpy-cross?

2) Should someone tell the maintainers of the pip package to update it?
Or better yet publish a new my-cross-v61 package, to leave peace of mind for those who still use the 1.19 firmware.

robert-hh, 2023-04-30:
I do not know how to get it unless you build it yourself. I attached the linux version that I have built. Maybe that works for now.
[mpy-cross-linux-v6.1.zip](https://github.com/micropython/micropython/files/11362181/mpy-cross-linux-v6.1.zip)

dpgeorge, 2023-05-01:
> 2\. Reading the docs, it seems the same MPY bytecode is compatible between versions.
>         If the version is 6 with 1.19 and 1.20 firmwares but the MPY files aren't compatible, I think it is misleading.
>         IMHO, breaking ABI / API changes have to increment the code version to 7.

This is the core of the issue: .mpy files have both a version and subversion and the subversion is used only when the file contains native code.  So v6 bytecode will run on 1.19 and 1.20, but v6.0 native code will only run on 1.19, you need to build v6.1 native code for 1.20.

This is described in the docs at http://docs.micropython.org/en/latest/reference/mpyfiles.html#versioning-and-compatibility-of-mpy-files

There is some code in that docs reference you can run on your device to see what version and sub-version you are using.  1.19 and 1.20 should be different.

If I run `mpy-cross --version` I get:
```
$ mpy-cross --version
MicroPython v1.20.0-29-ga39e2827b on 2023-05-01; mpy-cross emitting mpy v6.1
```

massimosala, 2023-05-01:
Hi folks!

I thank you all for the answers.

I just installed mpy-cross-v6 with pip (python 3.8).
If I run mpy-cross --version I get:

`MicroPython a327cfc on 2023-02-16; mpy-cross emitting **mpy v6**
`

It seems to me for the common man, using pip install, the problem isn't solved.

I open a ticket also here:
https://github.com/pybricks/python-mpy-cross/issues/28

I hope you two teams can talk a find a solution.

andrewleech, 2023-05-02:
I'm the author of the original [mpy-cross package on pypi](https://pypi.org/project/mpy-cross/), though not the "bytecode versioned" package you refer to above.

With the mpy-cross package you can simply install the version that matches your micropython, eg.
```
pip install mpy-cross==1.20.0
```
Then it should make mpy files that match your micropython, without needing to know what bytecode version it is behind the scenes.

massimosala, 2023-05-09:
> With the mpy-cross package you can simply install the version that matches your micropython, eg.
> 
> ```
> pip install mpy-cross==1.20
> ```

Thanks Andrew for your support.
Sorry I haven't tested your package, I have to open more issues and I have a tight development schedule.
My toolchain uses the other package, latest version:

```
pip install mpy-cross-v6.1
```

jonnor, 2024-07-03:
As mentioned above, one must use matching versions of .mpy files, and also the mpy-cross tool.

closed 2024-08-27 by projectgus