micropython/micropython #11374: mpremote: Add facility to specify location of config.py

peterhinch, 2023-04-29:
I am adapting [this project](https://github.com/Josverl/mpremote_config/blob/master/config.py) to add extra custom scripts. I wish to have `config.py` located on my server. The obvious way, a symlink in `~/.config/mpremote/` allows `config.py` to run Alas it runs in the directory of the symlink rather than in the directory containing the file. Consequently it doesn't pick up my scripts.

I currently export the envar `XDG_CONFIG_HOME` which works but is a hack. One option might be a MP-specific alternative envar, alternatively ensure that symlinks are followed.

esologic, 2024-07-08:
Yes I also ran into this recently. The docs aren't super clear that the `.config` file should be placed user's home directory - I thought that it should be in the project root ie: `/some/directory/my_project/.config/mpremote/config.py`. To me that makes more sense so one can have a `config.py` per project. 

I'd be interested in picking this up, as I agree setting `XDG_CONFIG_HOME` is a hack.

Josverl, 2025-09-05:
recently mpremote was updated to use [platformdirs](https://github.com/tox-dev/platformdirs) rather than custom logic.

For me the motivation was to make it work on windows, so I did not need to hack mpremote each version. In that change I also tried to clarify the docs.

While that does not explicitly provide a configuration option to override the location, at least it now it is following the default pythonic standard.