micropython/micropython #11378: Micropython reset loop after flashing

boredcoder411, 2023-04-30:
I recently bought a nodemcu amica, which uses the ESP8266 microcontroller. The problem I'm having is that when I start the chip up, and open puTTY, the chip just sends this message:
[image]
And then while restarting prints this for about half a second:
[image]

All this happens when flashing this binary:
esp8266-20230426-v1.20.0.bin
Does anyone know why this is?

robert-hh, 2023-04-30:
Try different flash modes when flashing the firmware. The default is DIO, but in many case it has to be DOUT
P.S.: This behavior has been reported gazillion times in the Forum and Discussions. So you may find more answers there.

boredcoder411, 2023-04-30:
On second thought, I will just use the nodemcu firmware. Thankyou for you time though!

closed 2023-04-30 by boredcoder411