micropython/micropython #11381: RasperryPi Pico W - Wifi issue with hidden SSID ?

Korhm, 2023-04-30:
Hello,

I have stability issues with my Raspi Pico W connected to a wifi network with SSID hidden
When I start the programm (with Thonny IDE run command, or ampy, or directly by plugin the Pico W to power supply), it does not always connect to wifi.

Most of the time after few seconds, I run into the error `network.STAT_CONNECT_FAIL` or `network.STAT_WRONG_PASSWORD` (but password is correct)
If it finally connect (after 3 or 4 physical or logical reset), it's not satable at all. ping command may more than 50% of the packet.

I did not find in the documentation of micropython (or of the Pico W) that hidden SSID may cause issue.

If I switch to a more classical Wifi netrowk (SSID not hidden), there is no issue

Is if something somebody already faced ?
Is it recommanded to use a not hidden SSID ?

I use the latestet micropython git branch (master - commit 867e4dd3dc9235974974fd5dab204371616d6f49) 
```
MicroPython v1.20.0-24-g867e4dd3d on 2023-04-28; Raspberry Pi Pico W with RP2040
```

with latest submodule versions
```
➜  micropython git:(master) git submodule status
 84f56af13292d8f32c40acbd949bde698ddd4507 lib/asf4 (remotes/origin/circuitpython)
 531cab9c278c947d268bd4c94ecab9153a961b43 lib/axtls (axtls-2.1.5-44-g531cab9)
 35aaec4418ad78628a3b935885dd189d41ce779b lib/berkeley-db-1.xx (heads/master-7-g35aaec4)
 1635e36d06821af8b61302509e91bfcc1ade84c4 lib/btstack (v1.4-1660-g1635e36d0)
 36144f94bee69ad5e7b46fef06ce0c2405bafee5 lib/cyw43-driver (v1.0.0)
 55bffd3a71cbeed2104cf30e7a39b641d8c1ff48 lib/fsp (v2.3.0-1-g55bffd3a)
 e9de7e35f2339598b16cbb375f9992643ed81209 lib/libffi (v3.2.1-146-ge9de7e3)
 5c5d513093075f7245ea522101b17c50aa579af2 lib/libhydrogen (5c5d513)
 6ca936f6b588cee702c638eee75c2436e6cf75de lib/lwip (STABLE-2_1_3_RELEASE)
 1bc2c9cb8b8fe4659bd94b8ebba5a4c02029b7fa lib/mbedtls (v2.16.10-61-g1bc2c9cb8)
 c113611765278b2fc8dcf8b2f2c3513b35a69b39 lib/micropython-lib (v0.1-1154-gc113611)
 42849560ba7906f023f61e5f7ff3709ba2c1dfca lib/mynewt-nimble (nimble_1_4_0_tag-2-g42849560)
 7a4c9d946cf1801771fc180acdbf7b878f270093 lib/nrfx (v2.0.0)
 fa5a554c7944d2a196626f8d3631e44943f9abcc lib/nxp_driver (fa5a554)
 f396d05f8252d4670d4ea05c8b7ac938ef0cd381 lib/pico-sdk (1.5.0-1-gf396d05)
 fdb1ec6879bc8b7bb54ded73bac04839003c5a07 lib/stm32lib (F0-1.9.0+F4-1.16.0+F7-1.7.0+G0-1.5.1+G4-1.3.0+H7-1.11.0+L0-1.11.2+L1-1.10.3+L4-1.17.0+WB-1.10.0+WL-1.1.0-35-gfdb1ec6)
 868f2bcda092b0b8d5f7ac55ffaef2c81316d35e lib/tinyusb (0.13.0-345-g868f2bcda)
 0803fc519ad7227e841287fb3638d6c8b2f111a1 lib/wiznet5k (W5500_PHY_LINK-92-g0803fc5)
```

I build a custom firmware by adding 2 python modules with a manifest. The commands to bulld are :
```
cd ports/rp2
make FROZEN_MANIFEST=/home/damien/dev/upiscine/boards/rp2_PCIO-W_manifest.py BOARD=PICO_W submodules
make FROZEN_MANIFEST=/home/damien/dev/upiscine/boards/rp2_PCIO-W_manifest.py BOARD=PICO_W clean
make FROZEN_MANIFEST=/home/damien/dev/upiscine/boards/rp2_PCIO-W_manifest.py BOARD=PICO_W
```
with `p2_PCIO-W_manifest.py`
```
include("$(PORT_DIR)/boards/PICO_W/manifest.py")
module("microdot.py", base_path="./libs/microdot/src")
module("microdot_asyncio.py", base_path="./libs/microdot/src")
```

Code to connect to network:
```python
import network
from secrets import secrets
from time import sleep
from rp2 import country

status_map = {
    network.STAT_IDLE: ('STAT_IDLE', 'no connection and no activity'),
    network.STAT_CONNECTING: ('STAT_CONNECTING', 'connecting in progress'),
    network.STAT_WRONG_PASSWORD: ('STAT_WRONG_PASSWORD', 'failed due to incorrect password'),
    network.STAT_NO_AP_FOUND: ('STAT_NO_AP_FOUND', 'failed because no access point replied'),
    network.STAT_CONNECT_FAIL: ('STAT_CONNECT_FAIL', 'failed due to other problems'),
    network.STAT_GOT_IP: ('STAT_GOT_IP', 'connection successful')
}

def wlan_connect():

    country('FR')

    ap_if = network.WLAN(network.AP_IF)
    ap_if.active(False)

    wlan = network.WLAN(network.STA_IF)
    wlan.deinit()

    if not wlan.isconnected():
        print('connecting to network {} ...'.format(secrets.get('wifi_sid')))
        wlan.active(True)
        wlan.connect(secrets.get('wifi_sid'), secrets.get('wifi_password'))
        while not wlan.isconnected():
            print('Error: {} ({})'.format(status_map.get(wlan.status()), wlan.status()))
            sleep(1)
    print('network config:', wlan.ifconfig())
```

Thank you!

Gadgetoid, 2023-05-02:
Are the hidden/not-hidden networks served by different hardware?

Have you tried `wlan.config(pm=0xa11140)`?

Flipje1955, 2023-05-05:
it may not be related but I experience that the WLAN interface returns inconsistent values specifically (but not only) on the "hidden" parameter. See: "RP2040: wlan.scan results disconnected from documentation and faulty #11048". In a recent test, I got values ranging from 0 to 5 whereas it should only be 0 or 1 (hidden/not_hidden).