micropython/micropython #11389: import returns an incomplete module if the last import attempt failed

DavidEGrayson, 2023-05-01:
Suppose you have a file named `error.py` with contents:

    raise Exception("This is a test exception.")

In CPython, you can run `import error` error multiple times and you get an exception every time, which is what I would exept:

```py
    Python 3.10.10 (main, Feb 10 2023, 19:20:41)  [GCC 12.2.0 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import error
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "E:/error.py", line 1, in <module>
    raise Exception("This is a test exception.")
Exception: This is a test exception.
>>> import error
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "E:/error.py", line 1, in <module>
    raise Exception("This is a test exception.")
Exception: This is a test exception.
```

In MicroPython, only the first attempt will raise an exception.  Subsequent attempts will not raise an error, which makes you think they have succeeded, but in fact they don't even run the code in the file and instead just return the module object that was generated by the previous import.  

```
>>> import error
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "error.py", line 1, in <module>
Exception: This is a test exception.
>>> import error
>>> import error
>>> error
<module 'error' from 'error.py'>
```

I tested this using on a Raspberry Pi Pico (RP2 port) running a custom-built version of MicroPython commit 867e4dd, and saw the same behavior with [rp2-pico-20230426-v1.20.0.uf2](https://micropython.org/resources/firmware/rp2-pico-20230426-v1.20.0.uf2).

This is a problem because it sometimes makes errors hard to debug.  If a library cannot be loaded because of a syntax error, and your error-handling code attempts to load that library again while showing the error, the second import will not work correctly and you will get some exception that is different from the actual syntax error that caused the root of the problem (i.e. an error that the imported library doesn't contain a name you were expecting it to contain).

I would like incompletely-loaded modules to be available in `sys.modules` so we can inspect them, see their variables, and look for the source of the error.  I think it might also be needed to support circular imports (see #10543).  However, an `import` statement should never let you import it: if the last import failed, it should try again.

DavidEGrayson, 2023-05-01:
OK, sure.  So I think `process_import_at_level` in `py/builtinimport.c` needs to somehow catch exceptions raised by `do_load`, and then remove the new module from the dictionary.  I'm not familiar enough with the code to know how to do this; is there some other place where Python exceptions are caught in C that we would look at?

dlech, 2023-05-01:
> is there some other place where Python exceptions are caught in C that I could look at?

Search for `nrl_push`.

> needs to somehow catch exceptions raised by `do_load`, and then remove the new module from the dictionary before re-raising them.

Or perhaps split up `mp_obj_new_module()` into two functions and don't add to `sys.modules` until later? I.e. change `MP_MAP_LOOKUP_ADD_IF_NOT_FOUND` to `MP_MAP_LOOKUP`, remove `el->value = MP_OBJ_FROM_PTR(o);` and add the module to `MP_STATE_VM(mp_loaded_modules_dict).map` later when no more exceptions could occur.

dpgeorge, 2023-05-01:
I think this is the same as (the very old) #967.

closed 2023-05-01 by DavidEGrayson