micropython/micropython #11404: RP2: machine.spi LSB first is unsupported (also SoftSPI)

peterhinch, 2023-05-03:
This arose [here](https://github.com/peterhinch/micropython-micro-gui/issues/31). The driver sets up a passed SPI instance as follows:
```py
    def __init__(self, spi, pincs, height=240, width=400, vcom=False):
        spi.init(baudrate=2_000_000, firstbit=machine.SPI.LSB)
        self._spi = spi
```
which fails with `NotImplementedError: LSB` (hard SPI) or `ValueError:firstbit must be MSB` (soft SPI).

The constant `LSB` does exist.
```py
>>> dir(SPI)
['__class__', '__name__', 'read', 'readinto', 'write', '__bases__', '__dict__', 'LSB', 'MSB', 'deinit', 'init', 'write_readinto']
```

robert-hh, 2024-07-09:
I made a PR once for LSB first with SoftSPI, but closed it since there were too many PRs open. I can re-open that PR, such that people can build their own firmware based on it.

robert-hh, 2024-07-09:
See PR #15436.

slabua, 2024-07-10:
@robert-hh thank you!

dpgeorge, 2024-07-12:
The RP2040 hardware SPI doesn't support LSB mode.  And it's quite difficult to synthesize LSB mode: eg if the driver transforms all the bytes before sending them out that won't be possible if the data to send is a read-only bytes object (and in ROM).

So I think the rp2 port just has to remain without SPI.LSB for hardware SPI.  Instead one would need to use software SPI if LSB mode is needed (and software SPI.LSB will be implemented in #15436).

slabua, 2024-07-12:
@dpgeorge i see thanks. Currently I am using hardware SPI by preprocessing the data thru a function that converts MSB to LSB. Not optimal but the only way I found.

dpgeorge, 2024-07-12:
> Currently I am using hardware SPI by preprocessing the data thru a function that converts MSB to LSB. Not optimal but the only way I found.

Once #15436 is merged I suggest you use `machine.SoftSPI` in LSB mode.

slabua, 2024-07-12:
@dpgeorge thanks, will do!

dpgeorge, 2024-07-12:
LSB for SoftSPI was implemented in ee1036023ef199636d96e70c6c7ed587ccaab92e

@peterhinch I think we can now close this issue.  I know rp2 hardware SPI still doesn't support LSB, but that's a hardware limitation that we can't really resolve.  Unless it automatically switches to software SPI when configured in LSB mode... the argument for that would be to make it transparently support LSB mode, instead of the user having to explicitly use SoftSPI when they want LSB.

mattytrentini, 2024-07-13:
> @peterhinch I think we can now close this issue. I know rp2 hardware SPI still doesn't support LSB, but that's a hardware limitation that we can't really resolve. Unless it automatically switches to software SPI when configured in LSB mode... 

What's the behaviour when you _try_ to use HW SPI and LSB on a port that doesn't have it? Ideally I think it should _not_ be a syntax error (ie firstbit should still be present) but an exception should be raised (ValueError?) that indicates that the option is not available for that port. I don't think it should automatically switch to SoftSPI...

dpgeorge, 2024-07-14:
> What's the behaviour when you _try_ to use HW SPI and LSB on a port that doesn't have it?

On rp2, if you try to construct a hardware SPI object in LSB mode, it raises a `NotImplementedError()`.

> I don't think it should automatically switch to SoftSPI...

As a counterpoint, the rp2 hardware I2C implementation switches automatically to software I2C for zero-length writes, eg `i2c.write(42, b"")`.  That's because the RP2040 hardware doesn't support zero-length I2C write transactions.  In this case it would be very annoying if it raised an exception for zero-length writes, because it would mean that you'd need to manually switch to software I2C if you wanted a zero-length write but wanted to take advantage of hardware I2C for all non-zero-length writes.

But in the case of SPI LSB mode, arguably you don't switch between LSB and MSB dynamically.  Rather, you pick one of those modes at the start and stick with it.  So if you need LSB then just use `machine.SoftSPI`.

closed 2024-07-18 by dpgeorge