micropython/micropython #11406: MicroPython allocates only 2MB of all flash memory on RP Pico 16MB

Hristo-Nikodimov-Nenkov, 2023-05-03:
I've just bought a Raspberry PI Pico (16MB) from AliExpress and tried both the 1.19.1 and the 1.20.0 versions of MicroPython.
With both it ( MicroPython ) allocates only 2MB of the available 16MB to the file system.

mattytrentini, 2023-05-03:
The [Raspberry Pi Pico](https://www.raspberrypi.com/products/raspberry-pi-pico/) only has 2MB flash. You need to match the board you bought with an appropriate board image. 

It's likely to be a fairly generic RP2040 micro with 16MB flash - try the [WeAct Studio 16MB](https://micropython.org/download/WEACTSTUDIO/) board image.

Flipje1955, 2023-05-03:
@mattytrentini: i share Hristo's interested in rp2040 with more (than 2MB) of flash. 

As with the reference to the WeAct that you include, there is no 16MB image accessible in the download pages. I addressed this aspect some time ago in the Q&A section (https://github.com/orgs/micropython/discussions/10964) but so far it has not had any suggestions yet. 
Having stated that, I appreciate that there is a (small) core of very motivated people doing great things for us and they cannot serve everybody. I know it is possible to build an image myself (but so far I miss the time and the confidence to dive into such an activity). Therefor, the RP2040 16MB boards that I have are for now in the pile of projects_to_look_into. Thanks !

mattytrentini, 2023-05-04:
> @mattytrentini: i share Hristo's interested in rp2040 with more (than 2MB) of flash.

The default build of the WeAct Studio _is_ for the 16MB version. (Or at least _should_ be, I can't test it right now...) 

I agree that it's not particularly clear since the other variants are labelled with the size of the flash but, [double-checking the files](https://github.com/micropython/micropython/tree/master/ports/rp2/boards/WEACTSTUDIO), the default firmware is indeed configured for 16MB. Please give it a try!

Building and publishing the other variants (2MB, 4MB, 8MB) to the [Download system](https://micropython.org/download/WEACTSTUDIO/) requires a small change from the core team. I'll ask again. For those that _can_ build their own images, the `BOARD_VARIANT` build option can be used to configure the flash size.

Flipje1955, 2023-05-04:
Hello Matty, you are correct, it is indeed the 16MB image and it works, also on an alternative outlet offered on AliExpress (SONGXIN LIGHT Store).
Thank you for response and follow up this. Much appreciated.

jonnor, 2024-09-01:
Problem looks to have been resolved, closing :)

closed 2024-09-01 by jonnor