micropython/micropython #11419: [ESP32 port] network_ppp does not manage pcb properly; can cause LWIP assertion or unmanaged pcb

madmax4889, 2023-05-04:
In handling of active(false), network_ppp.c ignores the return value of pppapi_free() and always sets self->pcb to NULL. However, pppapi_free() can fail if the PPP link is not currently dead, in which case the PCB remains allocated.
The implication of this is:
- The PCB is not freed and yet network_ppp.c no longer has a reference to it. The PCB is essentially 'leaked' at this point.
- Subsequent attempts to call active(true) will cause an LWIP assertion (which leads to a panic) due to the fact that the network interface we are trying to add already exists (due to the previously-unfreed pcb).

The scenanio above is exacerbated by the fact that:
- network_ppp.c has a 4000 ms timeout for the pppapi_close() call. However, if pppapi_close() was called while the LCP finite state machine had not yet set the phase of the PPP to DEAD, then LCP layer will try to send 2 TERM_REQs to its peer. If the peer is not answering, both of these TERM_REQs will timeout. In the version of LWIP, the timeout is configured for 6 seconds per attempt, for a total of 12 seconds. Therefore the LCP layer will require more than the 4 (or 4000ms) given by network_ppp.c. As a result, acitve(false) will timeout on pppapi_close(). Furthermore its call to pppapi_free() will fail and this failure will trigger the initial problem described above where network_ppp.c has NULLed its reference to the PCB eventhough the PCB hasn't been freed successfully in LWIP.

chihosin, 2023-07-11:
I has same problem with this issue, has some way to fix now?

[image]

chihosin, 2023-07-12:
I try to remove timeout check, check pppapi_free return code, and pin task to core 0, it working!

MCU: ESP32S3 P1N8

add some check for pppapi_free:
[image]

Pin to core 0:
[image]

Something wrong for PPP run in core1?