micropython/micropython #11422: mpremote should not limit name shortcuts to 1st 4 devices

bulletmark, 2023-05-05:
The mpremote help says you can use the following shortcuts:
```
  a0          connect to serial port "/dev/ttyACM0"
  a1          connect to serial port "/dev/ttyACM1"
  a2          connect to serial port "/dev/ttyACM2"
  a3          connect to serial port "/dev/ttyACM3"

  c0          connect to serial port "COM0"
  c1          connect to serial port "COM1"
  c2          connect to serial port "COM2"
  c3          connect to serial port "COM3"

  u0          connect to serial port "/dev/ttyUSB0"
  u1          connect to serial port "/dev/ttyUSB1"
  u2          connect to serial port "/dev/ttyUSB2"
  u3          connect to serial port "/dev/ttyUSB3"
```
That's good, but why limit them to 4? I had 5 devices connected so for the first 4 I could address easily, e.g. `mpremote u3` but the 5th device I had to specify the full device path (because `mpremote u4` is not accepted), i.e. I had to type `mpremote connect /dev/ttyUSB4`. Just seems like an unnecessary constraint and surely would be easy to fix.

This is using mpremote from master branch as of today (`v1.20.0-50-g786013d46`) .

massimosala, 2024-06-20:
Hi Mark

I found the minimal changes to support multi-byte characters and, as command-line options, all serial ports.
Apply the attached patches to the original source files.
Try it and let me know.

If it is OK, I will contact the  developer of _mpremote_
[mpremote_diff.zip](https://github.com/user-attachments/files/15919856/mpremote_diff.zip)

massimosala, 2024-06-20:
Ah, the package is this one:

> Name: mpremote
> Version: 1.23.0
> Summary: Tool for interacting remotely with MicroPython devices
> Home-page:
> Author:
> Author-email: Damien George <damien@micropython.org>
>

bulletmark, 2024-06-20:
@massimosala the `mpremote` code is hosted here in this repo (where I raised the issue). Sorry, but those changes would never be accepted.

I don't actually use `mpremote` directly, but rather I use my wrapper [`mpr`](https://github.com/bulletmark/mpr) and I added a change long ago to fix this issue in my wrapper. See the code [here](https://github.com/bulletmark/mpr/blob/213b56a2dba7aa5776bafb116ea30112c7a7dc91/mpr/mpr.py#L73).