micropython/micropython #11424: upip does not exist when importing from Micropython in ESP32

xITmasterx, 2023-05-05:
I am using the latest build of Micropython for ESP32, and I ran into this problem wherein I can't import upip at all. I have unflashed and reinstalled Micropython into ESP32, and I have reset it, but upip still does no exist when I tried to import it. Was there a problem in the latest build regarding this?

brianredbeard, 2023-05-05:
upip was deprecated a while ago - https://github.com/micropython/micropython/commit/924a3e03ec167c4417d89b531794c75ce5a631a3

brainelectronics, 2023-05-07:
I have currently a PR open to introduce `upip` again as installable package, https://github.com/micropython/micropython-lib/pull/632

jimmo, 2023-05-10:
upip has been unsupported for a long time. In v1.20 it was replaced by `mip`. See release notes https://github.com/micropython/micropython/releases/tag/v1.20.0 and documentation https://docs.micropython.org/en/latest/reference/packages.html#installing-packages-with-mip

closed 2023-05-10 by jimmo