micropython/micropython #11438: WebREPL on WIZnet W5500-EVB-Pico

detlevo, 2023-05-06:
The a.m. board contains both the webrepl modules (webrepl.py and webrepl_setup.py). I was able to set up webrepl. However, the next time I started the board, webrepl.start() failed because the setup_conn() function expects the network interfaces class network.WLAN to exist (see lines 104ff in webrepl.py). However, on this ethernet based board, no such class exists. It seems webrepl is only meant for WiFi enabled boards.

These are the possible resolutions:
1. Don't include the 'webrepl' module in the board firmware (i.e. don't include 'bundle-networking' in the boards manifest.py)
2. Modify webrepl.setup_conn() function such, that it works on this board as well.

robert-hh, 2023-05-06:
You can change webrepl.py to scan trough the interface list returned by network.route(). But some WiFi ports (e.g. CYW43) to not register their interfaces to network.route(), even if I recall faintly that this has been changed. Anyhow, then the scan loop in setup_conn() of webrepl.py would have to be changed from
```
    for i in (network.AP_IF, network.STA_IF):
        iface = network.WLAN(i)
        if iface.active():
            print("WebREPL server started on http://%s:%d/" % (iface.ifconfig()[0], port))
```

to
```
    for iface in network.route():
        if iface.active():
            print("WebREPL server started on http://%s:%d/" % (iface.ifconfig()[0], port))
```

detlevo, 2023-05-06:
Alternatively the whole loop could be simplified to 

print("WebREPL server started on port %s" % port)

Most of the time the user will not be connected via serial link in parallel. Why would he/she need to enable the WebREPL otherwise?