micropython/micropython #11439: Weird (mis)behaviour when passing kwarg along with *args

mirko, 2023-05-07:
When using the following library[1] I experienced a weird error:

Frozen module / copied into some importable path makes no difference.
I'm having the following interaction with the repl:

>>> from mp_modbus_master import modbus_rtu_master
>>> modbus_rtu_master()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/TMP/mp_modbus_master.py", line 513, in __init__
TypeError: unexpected keyword argument 'rtu'
>>>

Looking at the code, though, (https://github.com/eydam-prototyping/mp_modbus/blob/master/mp_modbus_master.py#L513) there's no kwarg `rtu`.
The only occurrence of the string "rtu" is the default value to the kwarg `ms_type` in that line.
That this string "rtu" is indeed the one seen as kwarg by mpy is confirmed by changing it to "XXX" - whereas mpy then complains about `TypeError: unexpected keyword argument 'XXX'`

Minimal test case showing this behaviour (thanks to @jimmo):

```
def a(x="y"):
    print(x)

def b(*args):
    a(x="z", *args)

b()
```

[1]https://github.com/eydam-prototyping/mp_modbus

mattytrentini, 2023-05-07:
Appears to be a problem introduced between v1.18 and v1.19:

```bash
❯ cat -n 11439.py
     1  def a(x="y"):
     2      print(x)
     3
     4  def b(*args):
     5      a(x="z", *args)
     6
     7  b()
     8

❯ docker run -ti --rm -v $(pwd):/code -w /code micropython/unix:v1.18 micropython-dev 11439.py
z

❯ docker run -ti --rm -v $(pwd):/code -w /code micropython/unix:v1.19 micropython-dev 11439.py
Traceback (most recent call last):
  File "11439.py", line 7, in <module>
  File "11439.py", line 5, in b
TypeError: unexpected keyword argument 'z'
```

dlech, 2023-05-07:
It appears we missed this case in https://github.com/micropython/micropython/pull/5807. Fix is on the way.

dhalbert, 2024-01-20:
A CircuitPython user discovered the same bug. It's also possible to cause a hard crash with similar code and a different argument.
https://github.com/adafruit/circuitpython/issues/8797
```
def f(a):
    pass

#f(a=2, *())     # TypeError: unexpected keyword argument ''
f(a=f, *())      # hard fault crash
```
Uncommenting the first call to `f()` causes an error similar to what was seen in previous posts.
Uncommenting the second call to `f()`, with the `a=f` argument, causes a hard crash.

dpgeorge, 2026-04-15:
This now raises a SyntaxError.