micropython/micropython #11440: ESP32-D0WD-3 SPIRAM does not boot with 1.20

brainelectronics, 2023-05-07:
## Info

- ESP32-WROVER-IE
- ESP32-D0WD-3
- 8 MB PSRAM
- 16 MB SPI
- esp32spiram-20230426-v1.20.0.bin

## Description

I'm using MicroPython `esp32spiram-20220117-v1.18` successfully since some time and now wanted to upgrade to 1.19.1 or even 1.20.0. I tried to flash `esp32spiram-20220618-v1.19.1.bin` (with and without SPIRAM) and `esp32spiram-20230426-v1.20.0.bin` (with and without SPIRAM) but with both firmware versions the device is no longer booting or even responding to mpremote REPL commands.

## Steps to reproduce
### Expected

Log flashing 1.18 and *doing* simple REPL interaction

```bash
(.venv) root@asdf:/Micropython/$ esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART erase_flash
esptool.py v3.1
Serial port /dev/tty.SLAB_USBtoUART
Connecting...
Failed to get PID of a device on /dev/tty.SLAB_USBtoUART, using standard reset sequence.
.....___
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:5a:6e:8c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.6s
Hard resetting via RTS pin...
(.venv) root@asdf:/Micropython/$
(.venv) root@asdf:/Micropython/$ esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 write_flash -z 0x1000 ../esp32spiram-20220117-v1.18.bin
esptool.py v3.1
Serial port /dev/tty.SLAB_USBtoUART
Connecting...
Failed to get PID of a device on /dev/tty.SLAB_USBtoUART, using standard reset sequence.
.....__
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:5a:6e:8c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x0018ffff...
Compressed 1631216 bytes to 1052559...
Wrote 1631216 bytes (1052559 compressed) at 0x00001000 in 25.2 seconds (effective 517.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

```bash
(.venv) root@asdf:/Micropython/$ mpremote connect id:eeb43b9f6179eb11b418fc83f771a14a ls
ls :
         139 boot.py
(.venv) root@asdf:/Micropython/$ mpremote connect id:eeb43b9f6179eb11b418fc83f771a14a repl
Connected to MicroPython at /dev/cu.SLAB_USBtoUART
Use Ctrl-] or Ctrl-x to exit this shell
>>>
MicroPython v1.18 on 2022-01-17; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import time
>>> time.time()
32
```

### Issue

Log flashing 1.20.0 and *trying* simple REPL interaction

```bash
(.venv) root@asdf:/Micropython/$ esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART erase_flash
esptool.py v3.1
Serial port /dev/tty.SLAB_USBtoUART
Connecting...
Failed to get PID of a device on /dev/tty.SLAB_USBtoUART, using standard reset sequence.
.....___
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:5a:6e:8c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.6s
Hard resetting via RTS pin...
(.venv) root@asdf:/Micropython/$
(.venv) root@asdf:/Micropython/$ esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 write_flash -z 0x1000 ../esp32spiram-20230426-v1.20.0.bin
esptool.py v3.1
Serial port /dev/tty.SLAB_USBtoUART
Connecting...
Failed to get PID of a device on /dev/tty.SLAB_USBtoUART, using standard reset sequence.
.....____
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:5a:6e:8c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00175fff...
Compressed 1527632 bytes to 990169...
Wrote 1527632 bytes (990169 compressed) at 0x00001000 in 24.0 seconds (effective 510.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

```bash
(.venv) root@asdf:/Micropython/$ mpremote connect id:eeb43b9f6179eb11b418fc83f771a14a ls
^CTraceback (most recent call last):
  File "/Micropython/.venv/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/Micropython/.venv/lib/python3.8/site-packages/mpremote/main.py", line 502, in main
    handler_func(state, args)
  File "/Micropython/.venv/lib/python3.8/site-packages/mpremote/commands.py", line 109, in do_filesystem
    state.ensure_raw_repl()
  File "/Micropython/.venv/lib/python3.8/site-packages/mpremote/main.py", line 444, in ensure_raw_repl
    self.pyb.enter_raw_repl(soft_reset=soft_reset)
  File "/Micropython/.venv/lib/python3.8/site-packages/mpremote/pyboard.py", line 360, in enter_raw_repl
    data = self.read_until(1, b"raw REPL; CTRL-B to exit\r\n>")
  File "/Micropython/.venv/lib/python3.8/site-packages/mpremote/pyboard.py", line 326, in read_until
    data = self.serial.read(min_num_bytes)
  File "/Micropython/.venv/lib/python3.8/site-packages/serial/serialposix.py", line 565, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
```

```bash
(.venv) root@asdf:/Micropython/$ mpremote connect id:eeb43b9f6179eb11b418fc83f771a14a repl
Connected to MicroPython at /dev/cu.SLAB_USBtoUART
Use Ctrl-] or Ctrl-x to exit this shell

```

```bash
(.venv) root@asdf:/Micropython/$ rshell -p /dev/tty.SLAB_USBtoUART
Using buffer-size of 32
Connecting to /dev/tty.SLAB_USBtoUART (buffer-size 32)...
Trying to connect to REPL

```

jonnor, 2024-09-02:
Sorry for no response. I am pretty sure others are using MicroPython on this chip. Are you certain that the board is actually switching out of the bootloader? I have observed several times that boards get stuck there, and in that case mpremote is unresponsible (but otherwise gives no error). Triggering the RESET pin on the board, or powering down/up again is a good way to check this.

Also, is the problem still present on latest MicroPython, version 1.23 or later?

jonnor, 2025-03-03:
Missing information to reproduce.

closed 2025-03-03 by jonnor