micropython/micropython #11449: os.sync not present on all platforms, how to be sure of filesystem consistency?

massimosala, 2023-05-09:
Current target: esp8266

On the official docs:
https://docs.micropython.org/en/latest/library/os.html

> os.sync()
> Sync all filesystems.
> 

It seems to me that there are no references in the esp8266 pages regarding its absence:
https://docs.micropython.org/en/latest/esp8266/quickref.html

I have a long experience on linux and servers, so I know the possible pitfalls about syncing the file-systems and sudden power failures.

However... two questions:
1) Why is os.sync not implemented on this target?
2) Are there other solutions ?

As a workaround: after writing to the files, I call time.sleep(5) before reset().
It seems to work... but it's a bad trick, not reliable.

Gadgetoid, 2023-05-10:
Looks like `sync()` is only implemented on stm32 and renesas-ra ports and is a no-op for filesystems other than FAT. I'd guess the renesas-ra port casually inherited it from stm32.

Documentation for `sync()` seems to have originated from an automated pass of comments in the C source - https://github.com/micropython/micropython/commit/88d3054ac072f9c73b0f3f045c59ba74f6730c1d - so it looks like a docs update is needed to clarify its availability and caveats.

1. The default filesystem for the esp8266 port is LittleFS, which has no concept of `sync()` (At least not in the flash-based MicroPython implementation) so it would just return `none` and you'd probably never know it was doing nothing :grimacing: I guess nobody has chosen to use FAT and asked for this feature. As you point out, it's not a panacea for power-loss data corruption anyway it would probably be more dangerous to include than not.

2. - assuming you're not using FAT anyway - have you tried `.flush()` on your output stream or file IO?:

```python
with open("file.txt", "w") as f:
    f.write("Something")
    f.flush()
```

Looks like we just fly by the seat of our pants on Pico: https://github.com/pimoroni/enviro/blob/337f7f1de4d7d73e3a2d0ccf111b8ae6fe7cdc30/enviro/__init__.py#L580C11-L583

jimmo, 2023-05-10:
os.sync exists for STM32 + Renesas where the filesystem is backed by internal flash with an in-RAM buffer. All other ports do not have a buffer, and writes are synchronous.

To your specific case on ESP8266 with LittleFS, as long as you are flushing (or closing) the file, there is no need to delay/sleep -- both flush and close are synchronous with respect to the underlying spiflash operations.

(Note that you do not need to explicitly flush before close, or at the end of the with block, as with the example above).

massimosala, 2023-05-18:
@jimmo Many thanks for your answers.

> (Note that you do not need to explicitly flush before close, or at the end of the with block, as with the example above).

If I understand: using with, in the __exit__ internals there is the right code to flush and close the file.
Correct ?

---

I think it will be super to add these details to the official docs.
Should I open a new "doc" ticket, citing this one?

jimmo, 2023-05-19:
> If I understand: using with statement, in its **exit** (internal method) there is the right code to flush and close the file. Correct ?

Yes. Specifically it closes the file (which as above, does a flush implicitly).

This is standard Python behavior, I'm not sure we need to add it explicitly to our docs, but if you can suggest a good place to put it then I guess there's no harm in adding clarity?

massimosala, 2023-05-19:
> os.sync exists for STM32 + Renesas where the filesystem is backed by internal flash with an in-RAM buffer. All other ports do not have a buffer, and writes are synchronous.

@Jimmo, I think this explanation has to be added to the official docs.
And it must be added in a general section, because those who don't use STM32 don't imagine having to look for these specifications in the docs of other micros.

I can't choose between these two pages:
> https://docs.micropython.org/en/latest/library/os.html
> https://docs.micropython.org/en/latest/reference/filesystem.html

os.sync is described in the first page, but this clarification is also useful when talking about the f.s.

---

> To your specific case on ESP8266 with LittleFS, as long as you are flushing (or closing) the file, there is no need to delay/sleep

I don't know what file system I have.
I installed the official firmware and use the builtin functions _open(filename, mode).read()_ or _open(filename, mode).write()_

> both flush and close are synchronous with respect to the underlying spiflash operations.

I thought the same, but I preferred to clarify, since @Gadgetoid had made an example with an explicit flush.

massimosala, 2023-05-19:
I hope the documentation gets improved.
Should I wait for feedback or can I close the issue ?

massimosala, 2024-06-20:
I made now a PR on the OS module doc.

Thanks to all

closed 2024-06-20 by massimosala