micropython/micropython #11458: RP2040 communication with MAX and processing ide

MishalJasmine, 2023-05-10:
### Discussed in https://github.com/orgs/micropython/discussions/11457

<div type='discussions-op-text'>

<sup>Originally posted by **MishalJasmine** May 10, 2023</sup>
In my project, RP2040 is connected to MAX3001 with which it communicates using SPI. The values are read back from MAX and plotted in a open source software called [processing](https://processing.org/). I have written the scripts in Micropython. But the issue is with the ports. 
Data is sent to the processing ide using UART. If I connect the port (COM4) in the processing ide, its gives no port available error since the same port (COM4)  is already connected in Thonny. 

Please let me know how to resolve this. </div>

Gawin-zippo, 2023-09-15:
Just put the code into main.py on your RP2040-based board, and use PySerial or some kind of stuff to fetch data from that COM Port instead of connecting your RP2040 board to REPL of IDE.

peterhinch, 2023-09-15:
This was resolved in Discussions.

closed 2023-09-15 by peterhinch