micropython/micropython #11464: Micropython fails to build with the latest GCC 13 release

stratakis, 2023-05-11:
Micropython on Fedora started failing with our latest updates to GCC 13.

Error:
```
../py/stackctrl.c: In function 'mp_stack_ctrl_init':
../py/stackctrl.c:32:32: error: storing the address of local variable 'stack_dummy' in 'mp_state_ctx.thread.stack_top' [-Werror=dangling-pointer=]
   32 |     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
../py/stackctrl.c:31:18: note: 'stack_dummy' declared here
   31 |     volatile int stack_dummy;
      |                  ^~~~~~~~~~~
In file included from ../py/runtime.h:29,
                 from ../py/stackctrl.c:27:
../py/mpstate.h:296:23: note: 'mp_state_ctx' declared here
  296 | extern mp_state_ctx_t mp_state_ctx;
      |                       ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../py/mkrules.mk:83: build/py/stackctrl.o] Error 1
```

Full build log attached.

[build.log](https://github.com/micropython/micropython/files/11447013/build.log)

dlech, 2023-05-11:
Should be fixed by https://github.com/micropython/micropython/pull/11402

stratakis, 2023-05-11:
Thanks for pointing it out. Trying out the fix.

stratakis, 2023-05-11:
Indeed the issue is fixed by #11402 . Sorry for the noise. Closing this.

closed 2023-05-11 by stratakis

dpgeorge, 2023-05-11:
If you need to build v1.20.0 without the error, try adding `CFLAGS_EXTRA="-Wno-dangling-pointer"` to the make invocation.