micropython/micropython #11473: Micropython freezing half way through print

MATTYGILO, 2023-05-11:
I have made code on an esp32 which communicates with firebase server and returns a response. The response is only a couple bytes and is a json. I am using the standard urequests script and its Response class which is resp. I can print and do anything I want with resp_json, however when I go to return resp_json the esp32 freezes half way through print as you can see

I also have stacks of free memory which is not the problem.

```
requests json() self.content: b'{"misc":{}}'
requests json() json_content: {'misc': {}}
firebase communicate new_dict: {'misc': {}}
Before
==================================================
Warning: communicate is not fully implemented
=================
```

```
resp = run_function(...)

    print("firebase mem_info:", micropython.mem_info())

    # TODO: Fix this very important bug
    # TODO: An error is therefore definitely occuring in resp.json()

    # TODO: Question 1: Does merely printing resp cause the error? No
    #       Question 2: Does merely printing resp.content cause the error? No
    #       Question 3: Does merely printing resp.status_code cause the error? No
    #       Question 4: Does merely printing resp.json() cause the error? No
    #       Question 5: Does storing resp.json() in a variable cause the error? No
    #       Question 6: Does returning resp.json() cause the error? Yes
    resp_json = resp.json()

    # Create a new dictionary object and copy the values
    # new_dict = copy_dict(resp_json)

    print("firebase communicate new_dict:", new_dict)

    # If success
    if True:  #int(resp.status_code) == 200:
        print("Before")
        time.sleep(10)
        print("="*50)
        print("Warning: communicate is not fully implemented")
        print("="*50)
        return new_dict
```

robert-hh, 2023-05-11:
What happens after `return new_dict`?  If the device is powered down or set to sleep, the print you have started may not finish. What happens if you add a sufficiently long sleep_ms() after the second print(), or sys.stdout.flush()?

peterhinch, 2023-05-12:
It's hard to assess this because it is only a fragment of code. Evidently it's the body of a function because it has a `return` statement, but the function definition and the context in which it is called are absent.

peterhinch, 2023-05-12:
@MATTYGILO I am moving this to discussions because there is no evidence that there is a MP bug here. If you can produce a minimal complete script which others can run to demonstrate a MP bug, please raise a new report.