micropython/micropython #11498: Guru Meditation Crash on ESP32 (MicroPython v1.20)

zachmoshe, 2023-05-14:
I have a ESP32 running MicroPython v1.20 and I'm recently seeing this crash happening sporadically (or at least I can't relate it to a specific function or event).

The ESP is using SD card via the SD/MMC 1-bit interface (`SDCard(slot=1, width=1)`), is connected to an I2S amplifier, uses ADC and a few leds (neopixel) and buttons. All these components I've worked with in the past, and using the same software drivers. I suspect it has something to do with memory but not sure and the error doesn't say anything specific (like the "couldn't allocate X bytes" that I normally get on these cases).

I've built Micropythom from the `v1.20.0` tag and added `ulab` (at commit `8585407`). My code itself is on the data partition and not embedded into the firmware if that makes any difference. 

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40120e10  PS      : 0x00060830  A0      : 0x80149120  A1      : 0x3ffce2a0  
A2      : 0x3ffee8c0  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffce35c  
A6      : 0x3ffed5d4  A7      : 0x00000008  A8      : 0x80120e10  A9      : 0x3ffce280  
A10     : 0x00000000  A11     : 0x49742400  A12     : 0x00000000  A13     : 0x3ffce360  
A14     : 0x3f4061c0  A15     : 0x0000000e  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000a  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x40120e0d:0x3ffce2a00x4014911d:0x3ffce2c0 0x400be95a:0x3ffce2e0  |<-CORRUPTED

ELF file SHA256: 953126fe619512f2

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4396
ho 0 tail 12 room 4
load:0x40078000,len:13884
load:0x40080400,len:3340
entry 0x40080618
```

It's a bit too cryptic for me, is anyone more familiar and can at least give a hint as for where the problem is? or what causes the crash? I'll try to eliminate problems from there but I'm currently completely in the dark...
I understand that I can't provide a small code to reproduce and the whole description is vague (also for me..) but I can't get any hint from the error message before the rebooting.

I'd love to provide more info or try things on the device.

Many thanks!

ChristianECooper, 2023-05-17:
Not sure if this helps, but the `LoadProhibited` code means there was a failure loading from memory.  The address being loaded when it failed is in `EXCVADDR`, which in your case `0x0a`.  That probably means that there was an attempt to read from a struct offset against a struct with a null pointer.

You can see what the panic errors mean here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html

jonnor, 2024-07-07:
Does the problem still occur on MicroPython 1.23 ? Does it also happen if ulab is not installed?

projectgus, 2024-08-27:
It wasn't here at the time, but we now have a guide to decoding esp32 backtraces out of MicroPython: https://github.com/micropython/micropython/wiki/ESP32-debugging

I'm going to close this out for now, but @zachmoshe if you see this again on 1.23 and you're able to decode the backtrace then feel free to submit another issue.

closed 2024-08-27 by projectgus