micropython/micropython #11511: uasyncio NameError: local variable referenced before assignment

bulletmark, 2023-05-16:
I see a MP language oddity (using ESPnow) which seems like a bug to me. If I run the trivial ESP32 code:

```python
import network
import uasyncio as asyncio
import aioespnow

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
e = aioespnow.AIOESPNow()
e.active(True)

async def main():
    async for mac, msg in e:
        print(mac, msg)

asyncio.run(main())
```
then I get a `NameError: local variable referenced before assignment` on line 11, the `async for ..` line. That global `e` variable is read-only in the `main()` function so this should not be happening. It works if I:

1. Pass `e` as an argument to `main(e)`, or
2. I add `global e` as first line in `main()`, or
3. Change to synchronous, i.e. use `espnow.ESPNow()`, i.e. don't use `asyncio`, or
4. I change `main()` to get rid of the `async for`:

    ```python
    async def main():
        while True:
            mac, msg = await e.airecv()
            print(mac, msg)
    ```

But none of this should be necessary. Just seems like a MP uasyncio bug.

I rebuilt MP from today's latest master (commit 3229791b60185faa47d87af80b6a8ccbb34d15e5) using ESP-IDF v4.4.4 to reproduce this issue.

jimmo, 2023-05-16:
@bulletmark Thanks for the report.

As you suspect this has nothing to do with aioespnow, but also not uasyncio. It's something with the core implementation of `async for`. Here's a minimal repro:

```py
class X:
    def __aiter__(self):
        return self

    async def __anext__(self):
        return 1

x = X()

async def main():
    async for a in x:
        print(a)

c = main()
c.send(None)
```

dpgeorge, 2023-05-16:
Thanks for the report.  I can confirm the issue as well.

The problem is with the compiler, the `compile_async_for_stmt()` function.  It's.... really broken!

Best work around at the moment is to pass in `e` as an argument to the function.  Using `global e` has other problems.

dpgeorge, 2023-05-19:
See fix in #11557.

dpgeorge, 2023-07-13:
Fixed by 606ec9bfb1cca262cf5b936db78924a609f9e73d

closed 2023-07-13 by dpgeorge