micropython/micropython #11533: Arduino Nano 33 BLE Sense not able to find USB serial connection after Flashing MicroPython

OKPersson, 2023-05-17:
Hey, I am trying to get *MicroPython* to work on my *Arduino Nano 33 BLE Sense* but am encountering an issue where the device isn't showing up in `/dev/` on either of my Mac or Linux computer.

The board works when flashing Arduino code via *Arduino IDE*, but after flashing *MicroPython* the only way to see the device via the command `ls /dev/` is to enter the bootloader mode, ie pressing the reset button twice.

I think it might be a regression of this: https://github.com/micropython/micropython/pull/8577#issuecomment-1189866928 Because I have had it working with that fix, but it's not working anymore.

The process of flashing is following the Arduino guide, running the *Arduino 33 BLE Sense* Example: `Nano33_updateBLandSoftDevice` using the *Arduino IDE 2.1.0*, opening the serial monitor and pressing y for it to update the soft device on the microcontroller. Then I use either the *BOSSAC* variant that is included with the *Arduino IDE* or https://github.com/arduino/BOSSA/releases/tag/1.9.1-arduino2

Firmware I have tried recently was `v1.20.0(2023-04-26)` and `v1.20.0-68-g3229791b6 (2023-05-11)`.
Guide followed: https://docs.arduino.cc/tutorials/generic/micropython-installation

Things that I haven't been able to try is to re-flash the firmware to the *Nano*, as the firmware/soft device sketch for *Arduino* seems to recognize that the firmware is already the newest one (Unsure if it checks firmware hash to see if it's broken).

It might be a hardware bug on my device or it could possibly be a bug, hopefully someone smarter than me can give some insights if I am doing something wrong. I only have one of the microcontroller so I cannot do a side by side test. I haven't been able to find anyone else having the same issue on the internet so it might just be my device that is malfunctioning, in that case I can just use *Arduino IDE* instead, but I want to try out *MicroPython* to learn more about it.

robert-hh, 2023-05-18:
Just try to reload the firmware, taking the latest daily build from https://micropython.org/downloa/arduino_nano_33_ble_sense/, and using the firmware update instrauctions from that site:
- double push reset
- run the command:
`bossac -e -w --offset=0x16000 --port=ttyACM0 -i -d -U -R arduino_nano_33_ble_sense-20230511-unstable-v1.20.0-68-g3229791b6.bin`
And that works. The bin file name may have to be adapted.

OKPersson, 2023-05-18:
I did try to to do that, the device is visible in `/dev/tty.usbmodem14214201` after flashing the softdevice, and using the *Arduino IDE* with serial monitor you can install the new softdevice when the device isn't in bootloader mode.
I tried it following your input as well, not just like 20 times before.

However, as soon as *MicroPython* gets flashed it will not show up in `/dev/`.

There is not error messages from BOSSAC, it writes the buffer, then reset()

Either my device has some hardware fault or software fault, or this isn't working. It could be a broken Python script on the device perhaps, as I have had *MicroPython* working during last summer. Is there a way to overwrite the usable flash storage so that older sketches might be overwritten?

robert-hh, 2023-05-18:
Do you see the progress messages of bossac? The -e option already erase the flash, see `bossac --help`. Maybe the softdevice previously used has been overwritten. The arduino nano requires s140 v6.1.1. So maybe you can revert that to this version.

OKPersson, 2023-05-18:
Yes, this is the output in the terminal:

```
./bossac -e -w --offset=0x16000 --port=tty.usbmodem14214201 -i -d -U -R arduino_nano_33_ble_sense-20230518-unstable-v1.20.0-69-gf9958417d.bin

Set binary mode
version()=Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Connected at 921600 baud
identifyChip()=nRF52840-QIAA
write(addr=0,size=0x34)
writeWord(addr=0x30,value=0x400)
writeWord(addr=0x20,value=0)
version()=Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Erase flash
chipErase(addr=0x16000)

Done in 0.001 seconds
Write 274316 bytes to flash (67 pages)
[                              ] 0% (0/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x16000, size=0x1000)
write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x17000, size=0x1000)
write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x18000, size=0x1000)
[=                             ] 4% (3/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x19000, size=0x1000)
[=                             ] 5% (4/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x1a000, size=0x1000)
[==                            ] 7% (5/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x1b000, size=0x1000)
[==                            ] 8% (6/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x1c000, size=0x1000)
[===                           ] 10% (7/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x1d000, size=0x1000)
[===                           ] 11% (8/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x1e000, size=0x1000)
[====                          ] 13% (9/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x1f000, size=0x1000)
[====                          ] 14% (10/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x20000, size=0x1000)
[====                          ] 16% (11/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x21000, size=0x1000)
[=====                         ] 17% (12/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x22000, size=0x1000)
[=====                         ] 19% (13/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x23000, size=0x1000)
[======                        ] 20% (14/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x24000, size=0x1000)
[======                        ] 22% (15/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x25000, size=0x1000)
[=======                       ] 23% (16/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x26000, size=0x1000)
[=======                       ] 25% (17/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x27000, size=0x1000)
[========                      ] 26% (18/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x28000, size=0x1000)
[========                      ] 28% (19/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x29000, size=0x1000)
[========                      ] 29% (20/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x2a000, size=0x1000)
[... 73 lines cut ...]
writeBuffer(scr_addr=0x34, dst_addr=0x4f000, size=0x1000)
[=========================     ] 86% (58/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x50000, size=0x1000)
[==========================    ] 88% (59/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x51000, size=0x1000)
[==========================    ] 89% (60/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x52000, size=0x1000)
[===========================   ] 91% (61/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x53000, size=0x1000)
[===========================   ] 92% (62/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x54000, size=0x1000)
[============================  ] 94% (63/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x55000, size=0x1000)
[============================  ] 95% (64/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x56000, size=0x1000)
[============================= ] 97% (65/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x57000, size=0x1000)
[============================= ] 98% (66/67 pages)write(addr=0x34,size=0x1000)
writeBuffer(scr_addr=0x34, dst_addr=0x58000, size=0x1000)
[==============================] 100% (67/67 pages)
Done in 10.703 seconds
reset()
```

I honestly don't know what is wrong, but I am almost certain that it's some hardware level fault now, at the same time it works as it should when flashing Arduino Sketches via the IDE.

iabdalkader, 2023-05-22:
Didn't you manage to fix that issue the last time ? Anyway, I can't reproduce this:

```bash
make -j12 BOARD=arduino_nano_33_ble_sense SD=s140 FLASHER=bossac deploy

MPY: soft reboot
MicroPython v1.20.0-74-gff08f1efa-dirty on 2023-05-22; Arduino Nano 33 BLE Sense with NRF52840
Type "help()" for more information.
>>> 
```

Atzingen, 2023-06-27:
Having the same proble here with two boards out of 5. 
Flash seams ok with boosac or the openMV ide and them it does not show as COM port on windows or /dev/tty*something* on linux. 
Since the flash process worked on some boards and not on others, is is probably a hardware problem as @OKPersson suggested.

OKPersson, 2024-02-17:
I did not, I take it it's either been rectified in newer revisions of the Nano33 BLE Sense, a hardware fault or user error sadly.

Perhaps it would be fixable if you could force reinstall the firmware/softdevice as I guess that it has to do with that. Currently the Arduino sketch doesn't allow that as far as I know.

As my device was used to test the early release of MicroPython for the Nano33 BLE Sense before they implemented https://github.com/micropython/micropython/pull/8577#issuecomment-1189870600 

However as more people seem to have the same or similar problems there could be a larger issue at hand. 

Which revision of the board do you have? Mine is a rev 1 ABX00035 .

It's been a while since I tried, maybe the issue has been fixed with newer software releases. I see that there is a MicroPython installer program now.

OKPersson, 2024-02-17:
> I did not, I take it it's either been rectified in newer revisions of the Nano33 BLE Sense, a hardware fault or user error sadly.
> 
> Perhaps it would be fixable if you could force reinstall the firmware/softdevice as I guess that it has to do with that. Currently the Arduino sketch doesn't allow that as far as I know.
> 
> As my device was used to test the early release of MicroPython for the Nano33 BLE Sense before they implemented [#8577 (comment)](https://github.com/micropython/micropython/pull/8577#issuecomment-1189870600)
> 
> However as more people seem to have the same or similar problems there could be a larger issue at hand.
> 
> Which revision of the board do you have? Mine is a rev 1 ABX00035 .
> 
> It's been a while since I tried, maybe the issue has been fixed with newer software releases. I see that there is a MicroPython installer program now.

Well, the installer program did not help sadly, installation got timed out when waiting for the device after flashing. Will try to do a more in depth check later.

saurabh-2905, 2024-02-18:
@OKPersson Thanks for the reply. Mine is Rev 2. Actually, I have purchased 10 of those pieces, so when the first one broke, I tried with the other piece, and for that it works fine till now. 

Yes, the micropython installer times out or throws some error. I am guessing it has something to do with softdevice installation, because the device is not bricked as it is able to run Arduino scripts. 

For the broken piece, I plan to re-flash the bootloader itself. Let's see if I can get my hands on the debugger soon to do this. I hope this solves the problem.

OKPersson, 2024-02-21:
@saurabh-2905 Please update here if you manage to fix the issue. Hopefully it's a semi easy fix, but still I wonder what the main cause of it is, why it's happening to some boards and some are fine I mean.

jonnor, 2024-09-01:
Issue not reproducible, so I think we should close this - since there is not much we can do. Can be reopened if able to provide details on how to reproduce.

closed 2024-09-01 by jonnor