micropython/micropython #11536: Pico W : WebREPL : Ctrl-C not working

HopWorks, 2023-05-18:
### Discussed in https://github.com/orgs/micropython/discussions/11532

<div type='discussions-op-text'>

<sup>Originally posted by **villeneuve** May 17, 2023</sup>
Hi!
With latest 1.20 (also tested with today nightly build) pressing Ctrl-C in WebREPL on a pico W that is running a program has no effect.
Ctrl-C works in normal REPL but not in WebREPL
On ESP8266 and ESP32 (latest 1.20 as well) it works as expected in WebREPL
And also (on the pico W) webrepl_setup exits on error because there is no boot.py file.
We can just create an empty one and it works but this isn't obvious for beginers (boot.py exists on ESP* ports) 
</div>

HopWorks, 2023-05-18:
I have the same issue as the OP on the included discussion, plus, I get nothing in the webREPL control on my Chrome web browser. My prints are not showing up, and when my program crashed for whatever reason, the output of that crash did not show either. All I got was disconnected.

gmos, 2023-06-23:
Also have the first issue on the Pico.  After having entered the password in the webrepl, nothing seems to happen.  Only when the running program ends and MO drops into the repl, the thing comes to live.  
Aborting the running program with ^C can only be done from the usbserial.
Noticed this 1st time running under asyncio, but same result in minimal main just doing time.sleep after webrepl set-up.

also when you disconnect, while the program is still running, MP thinks the connection is still in tact and refuses a new one.

felixdoerre, 2024-01-28:
btw: #9297 was resolved, so this issue should be resolved as well.