micropython/micropython #11543: micropyton.native and micropython.viper aren't defined

massimosala, 2023-05-18:
System:
> MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266

I am writing code optimized on python with numba and on micropython with viper.

The natural coding should be:

```
opt = micropython.viper
@opt
def my_func(...):
    ...

```

and

```
opt = numba.jit
@opt
def my_func(...):
    ...

```

Test:
```
import micropython
micropython.native
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'native'

```

In effect, _dir(micropython)_ doesn't show _native_ and _viper_

Temporary solution: in python with numba I create these stubs,  so I always use the decorator _micropython.native_
```
class micropython :
	native = lambda x: x
	def viper(func, *args, **kwargs) :
		return	numba.jit(func, *args, **kwargs, nopython = True)
```

It it ugly.
Please, is it possible to add the _native_ and _viper_ to the module?

jimmo, 2023-05-19:
I don't think this can be fixed in an easy way that doesn't add code size. Unfortunately it's a cost that everyone pays but almost nobody needs.

The issue isn't that the `.native` and `.viper` attributes don't exist on the `micropython` module, rather that this decorator is applied by the compiler (i.e. it changes the behavior of the compiler as it does the code generation for that function). In order for the code snippet you provided to work, the compiler would have to know what the `opt` variable was.

Your workaround for numba seems reasonable though.

massimosala, 2023-05-19:
I understand, it's not worth it.
Thanks, Massimo

closed 2023-05-19 by massimosala