micropython/micropython #11544: user defined functions, get function name

massimosala, 2023-05-18:
> MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266

Not specific to this port.

In Python for debugging, expecially in some cases like with nested decorators, I have to get the name of the functions (passed as arguments) with _func.__name___

Is it possible to have this attribute?
How much memory will it consume?

jimmo, 2023-05-19:
> Not specific to this port.

Unfortunately, in this specific case, it is!

Support for `func.__name__` is controlled by the config option `MICROPY_PY_FUNCTION_ATTRS`, which is disabled by default on ESP8266 (and other smaller-flash configurations). You can compile MicroPython yourself with this enabled.

Historically the ESP8266 has several features disabled because the original modules had 512kiB flash.

It might be worth considering enabling this on the 2MiB (and maybe 1MiB) flash size boards though?

massimosala, 2023-05-19:
Hi!
IMHO if it takes a few bytes per function, I think is valuable.
Are we talking about flash, where it is the firmware? not the free and precious heap.

jonnor, 2024-09-01:
If anyone volunteers to make a MR to enable on 2MiB (and maybe 1MiB) ports, that will probably be considered.
Otherwise I am proposing to close this issue, as it is possible to rebuild yourself via the standard instructions.

closed 2025-02-04 by projectgus