micropython/micropython #11583: missing ubinascii module in NUCLEO-G474RE binary

jancotipsarevic, 2023-05-21:
The ubinascii module is not present in the NUCLEO-G474RE so it is causing problems using ampy or rshell.

Thank you very much

dpgeorge, 2023-05-22:
The G474 is a bit tight on flash (352k for firmware, 160k for filesystem) so some features like `binascii` have been disabled.

I would suggest using `mpremote` https://pypi.org/project/mpremote/ to access the device from the command line.  It doesn't need `binascii`.

(Note that the G4 does have `bytes.fromhex()`, which is the same as `binascii.unhexlify()`, and is all that `rshell` needs to work.  I think it would be good for `rshell` to fallback to `bytes.fromhex()` if `binascii` is not available.)

jonnor, 2024-08-24:
As described above this is intentional, and solutions are available.

closed 2024-08-24 by robert-hh