micropython/micropython #11592: mpremote: how to send keyboard interrupt to a running coro task

tferrin, 2023-05-23:
This may not be a bug but rather just my lack of understanding or some other "user error."  I've been using mpremote on macOS for ~6 months now with very satisfactory results (requires defining an alias plus manipulating port identifiers in my .bashrc file, but once that's done mpremote works great!). But for the life of me I can't figure out how to send a KeyboardInterrupt signal when I'm debugging my scripts, which is especially needed when developing coro tasks. Here's what I'm talking about: if I put the following code at the end of a script I _never_ get the exception/finally clauses to run. There's no combination of ^C, ^D, ^], ... that causes the keyboard interrupt/finally code to get run. So am I missing something critical in the documentation or is this just a limitation of mpremote?

> try:
>     asyncio.run(main())
> except KeyboardInterrupt:
>     print("kb interrupt")
> finally:
>     asyncio.new_event_loop()    # clear any retained coro state

jimmo, 2023-05-23:
@tferrin Can you provide a more complete example including how you're using `mpremote`.

For example, if you're using `mpremote run` then the Ctrl-C is handled entirely locally and is not sent to the device -- it just stops mpremote waiting for the script to finish.

If you're in what mpremote calls the repl mode, then Ctrl-C will be sent to the running program, and in this case (unless I'm missing something about your example) you should see the KeyboardInterrupt.

e.g.

main.py:
```py
import uasyncio as asyncio

async def main():
    try:
        while True:
            await asyncio.sleep(1)
    except asyncio.CancelledError:
        print("cancelled")
    finally:
        print("main interrupted")

try:
    asyncio.run(main())
except KeyboardInterrupt:
    print("kb interrupt")
finally:
    print("finally")
```

Running with mpremote:
```
$ mpremote cp ~/mpy/async_kbdintr.py :main.py
$ mpremote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell

>>> 
>>>                                                # <-- hit Ctrl-D here to trigger soft reset (to run main.py)
MPY: sync filesystems
MPY: soft reboot                        # <--- hit Ctrl-C here
kb interrupt
finally
MicroPython v1.20.0-24-g867e4dd3d-dirty on 2023-04-28; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>
```

Note that you do not see the finally from main -- as far as the task is concerned, it's still waiting for the loop to keep running it.

All that said -- I do think there is a compatibility bug here. In CPython, it appears that asyncio.run handles KeyboardInterrupt and shuts down and cancels the running task. So you _don't_ see the KeyboardInterrupt, but the main task is cancelled and both finally blocks execute.

```
$ python ~/mpy/async_kbdintr.py
^Ccancelled
main interrupted
finally
```

peterhinch, 2023-05-23:
@tferrin The behaviour of keyboard interrupts in `uasyncio` is also discussed [here](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#512-keyboard-interrupts).

tferrin, 2023-05-23:
@jimmo and @peterhinch, thank you both for the insight and link to the relevant tutorial. I had tried mpremote run and also copying the code to the board and running it with mpremote exec 'import foobar', but now see that in both cases the Ctrl-C is only handled locally and never sent to the board. But if I copy my code to main.py and then do a soft reset to start it running, Ctrl-C causes an exception as expected. So I just need to change my approach to testing my code. Easy!

closed 2023-05-23 by tferrin