micropython/micropython #11594: Import causes object id changes in other modules

fragmuffin, 2023-05-23:
## Description

I discovered that, in a very specific case, importing can disrupt object ids after they've already been imported and used.

## Replicating Fault

I've attached a minimal subset of our code-base recreating the fault: [`bug-report.tar.gz`](https://github.com/micropython/micropython/files/11539260/bug-report.tar.gz)

```
$ tree
.
├── lib
│   └── uboot
│       ├── forwarder.py
│       └── services
│           ├── __init__.py
│           └── webterm
│               ├── connect.py
│               └── __init__.py
├── test.py
└── test.sh
```

The `test.py` imports a submodule (`webterm`), and references an object it contains (`webterm.connect`).

```python
...
from uboot.forwarder import webterm
function = type(lambda: None)

def test():
    if isinstance(webterm.connect, function):
        print("✅ Passed : {}".format(type(webterm.connect)))
    else:
        print("❌ Failed : {}".format(type(webterm.connect)))

test()
from uboot.services.webterm.connect import FOO  # <-- causes webterm.connect to change
test()
```

After importing a parallel object (referencing it more directly), the original value changes.

I believe the root cause of this may be due to the content of `uboot.services.webterm.__init__.py`:

```python
from .connect import connect
```

### Test Results (cPython vs micropython)

Note: This behaviour is not shared by cPython (not in v3.10, nor in v3.4 (micropython's original basis))

```
$ ./test.sh 
-------- sys.version=3.4.10 (default, Mar 20 2019, 00:52:37)  [GCC 6.3.0 20170516]
✅ Passed : <class 'function'>
✅ Passed : <class 'function'>
-------- sys.version=3.10.11 (main, May  3 2023, 09:20:51) [GCC 10.2.1 20210110]
✅ Passed : <class 'function'>
✅ Passed : <class 'function'>
-------- sys.version=3.4.0      ... results from micropython v1.18
✅ Passed : <class 'function'>
❌ Failed : <class 'module'>
-------- sys.version=3.4.0; MicroPython v1.20.0 on 2023-05-14
✅ Passed : <class 'function'>
❌ Failed : <class 'module'>
```

jimmo, 2023-05-23:
Thanks for the report -- here is a simpler demonstration:

```
$ tree .
.
├── mod
│   ├── foo.py
│   └── __init__.py
└── test.py

2 directories, 3 files
$ cat test.py 
import mod
print(mod.foo)
import mod.foo
print(mod.foo)
$ cat mod/foo.py 
def foo():
    print("foo")
$ cat mod/__init__.py 
from .foo import foo
$ python test.py 
<function foo at 0x7f73c4aa1300>
<function foo at 0x7f73c4aa1300>
$ micropython test.py 
<function foo at 0x7f5d4365b880>
<module 'mod.foo' from '/home/jimmo/mpy/bug-11594/mod/foo.py'>
```

Will have to look up the exact semantics from CPython here, but I'm guessing it's got something to do with how we build the module structure (i.e. storing `foo` on `mod`). From some quick tests I don't think it's quite as simple as "don't replace anything that's already there", but hopefully it's not too complicated.

jimmo, 2023-05-23:
Sorry missed the obvious explanation, I actually already fixed this bug in https://github.com/micropython/micropython/pull/11456, hopefully that will be merged soon.

I will add this as a test case to that PR. Thanks @fragmuffin

dpgeorge, 2023-05-23:
Thanks for the comprehensive bug report!

fragmuffin, 2023-05-24:
> @jimmo : Thanks for the report -- here is a simpler demonstration [🔗](#issuecomment-1558611070)

That's *so much* simpler, thank you.

> @dpgeorge : Thanks for the comprehensive bug report! [🔗](#issuecomment-1559121956)

My pleasure.

> @jimmo : I actually already fixed this bug [🔗](#issuecomment-1558919071)

That's amazing! :tada: thanks @jimmo !

My apologies for not testing on that PR.... but I don't think I would have known to try :shrug:
I'm glad this has contributed a test-case though :muscle:

jimmo, 2023-06-02:
Fixed by https://github.com/micropython/micropython/pull/11456

closed 2023-06-02 by jimmo