micropython/micropython #11617: builtin find isn't working as in Python

massimosala, 2023-05-25:
On _Python 3.8.10_ these work

```
>>> b'ciaob'.find(98)
4

>>> bytearray("ciaob", "ascii").find(98)
4
```

On _MicroPython v1.20.0_ don't work
```
>>> b'ciaob'.find(98)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'int' object to str implicitly

>>> bytearray("ciaob", "ascii").find(98)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'int' object to str implicitly
```

In regular Pythoh _find_ accepts and _int_ as needle when working with raw bytes as haystack.
IMHO this operation is very common coding on micros.

Rationale:
- avoid to do type conversions
- keep consistency with Python

Could you please add this feature to _find_ ?

robert-hh, 2023-05-25:
MicroPyhton does not claim compatibility to Python 3.8. It's based on Version 3.4 with some extensions from later versions. And besides that, it's Micro.

massimosala, 2023-05-25:
> MicroPyhton does not claim compatibility to Python 3.8. It's based on Version 3.4 with some extensions from later versions. And besides that, it's Micro.

I know, I know.

IMHO this is a small feature, easy to implement and will ease porting of code from Python to Micropython.

massimosala, 2023-05-30:
> > IMHO this is a small feature, easy to implement and will ease porting of code from Python to Micropython.
> 
> Just to make this concrete rather that theoretical, I implemented it to find out. See #11631
> 
> It is "small", but it's not trivial. +72 bytes may not sound like much but this adds up, so needs to be carefully considered.

Hi @jimmo !

Thanks for your efforts.

I am reading now pull #11631, I will comment there.

closed 2023-05-30 by massimosala