micropython/micropython #11618: integers: add support for bit_length() and bit_set()

massimosala, 2023-05-25:
On Python, from version 2.8, _int_ has the method _bit_length_

While developing on different versions of Python, I use this:

```
def bit_length(x):
	if hasattr(x, "bit_length") :
		return	x.bit_length()
	i = 0
	while x :
		x = x >> 1
		i += 1
	return	i

assert	bit_length(0) == 0
assert	bit_length(1) == 1
assert	bit_length((1 << 64) - 1) == 64
```

Python doesn't have a built-in _bit_set_, but it is very useful:

```
def bit_set(x) :
	i = 0
	while x :
		x = x & (x - 1)
		i += 1
	return	i

assert	bit_set(0) == 0
assert	bit_set(1) == 1
assert	bit_set(1 << 64) == 1
```

The two algorithms are efficient, simple and straightforward to code in C.

Could you please implement them on the _int_ type ?

robert-hh, 2023-05-25:
> Do I need special privileges to open a PR?

No.

> Please, is there a How to on PR?

It's a github thing. See https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

robert-hh, 2023-05-25:
> Python doesn't have a built-in bit_set,

Your code example seems to perform the `bit_count()` operation.

massimosala, 2023-05-25:
> we can see how large the implementation is

BTW, I think the implemetation will be tiny.
It isn't rocket science, just basic functions :-)

I'm willing to try to code the two functions, I've been developing in C (and other languages, on servers and on micros) for 30 years.
I need a little help getting started.

robert-hh, 2023-05-25:
> Iirc for bit_length there already exists an old pull request.

Sure? Do you know the PR number? A search does not reveal it.

dlech, 2023-05-25:
It looks like it is attached to this issue, but no PR was made: https://github.com/micropython/micropython/issues/4065

rkompass, 2023-05-25:
I meant that.

closed 2023-05-30 by massimosala