micropython/micropython #11620: os.sync availability

massimosala, 2023-05-25:
Hi

Please see
https://github.com/micropython/micropython/issues/11449

@jimmo wrote:

> os.sync exists for STM32 + Renesas where the filesystem is backed by internal flash with an in-RAM buffer. All other ports do not have a buffer, and writes are synchronous.

I've wasted time reading docs, finding a function, conding with it... and finding out it doesn't work because it's not implemented!

I kindly but firmly ask you to remove this function on all micros where it is a No Operation.
It will save confusion and frustration for all other developers as well.

Leaving it as NOP is terrible, because programmers think it's necessary and account for a file-system that requires sync... when it doesn't.

I've been trying to set up a PR, but it seems I don't have the privilege.

jimmo, 2023-05-26:
Please don't raise duplicate issues. Closing as duplicate of #11449

> I kindly but firmly ask you to remove this function on all micros where it is a No Operation.
> Leaving it as NOP is terrible,

I'm not sure I follow. This function does only exist on STM32 and Renesas ports. There is currently no port where it's a no-op.

That said, I would argue that we _should_ make it available on all ports, and leave it as a no-op on ports where it doesn't need to do anything. This means that someone who needs to ensure the filesystem is synchronized can write code that works everywhere. i.e. if your code expects the disk cache to be flushed, then you shouldn't have to care about the implementation, just call the function. If there's nothing to do (because there is no cache) then that's fine.

However, with MicroPython we're looking for every possible way to save code space, so not including a function is obviously cheaper. I'm not sure what the right answer is here.

The misleading part here is that the _documentation_ should mention that it's only available on stm32/renesas.

> I've been trying to set up a PR, but it seems I don't have the privilege.

See #11619, there are no restrictions on making a PR.

massimosala, 2023-05-30:
> I'm not sure I follow. This function does only exist on STM32 and Renesas ports. There is currently no port where it's a no-op.
...
> The misleading part here is that the _documentation_ should mention that it's only available on stm32/renesas.

@jimmo Thanks for your patience.

I try to follow your reasoning and simplify my requests.

1) Documentation

The docs should be updated.

Reading this  page it seems _sync_ is available on all ports:
_https://docs.micropython.org/en/latest/library/os.html_

And here it isn't listed as not implemented:
_https://docs.micropython.org/en/latest/esp8266/quickref.html_

---

> However, with MicroPython we're looking for every possible way to save code space, so not including a function is obviously cheaper. I'm not sure what the right answer is here.

I agree with your doubts.
It is convenient to write routines once using _sync_ and usable on several micros.

On micros like esp8266, how much flash space is required to implement _sync_ like a NOP? I think very few bytes.

I stress that in some cases, even if only for the client's specifications, it is necessary to know if the underlying file-system requires _sync_ or not.

My proposal:
- implement _sync_ on all ports
- where it does syncing, it returns True
- where it is a NOP, it returns False

IMHO it seems rational and useful.
With the same function it is possible to know its behaviour.

What do you think?

robert-hh, 2023-05-30:
> where it does syncing, it returns True
where it is a NOP, it returns False

For compatibility with CPython, os.sync() must not return anything. A trial implementation of a `NOP` implementation takes 40 bytes of code. That's does not seem a lot, but on some boards this is just wasted space. But these ports are usually anyhow restricted, so a missing os.sync() may be the least problem.

massimosala, 2023-05-30:
Hi Jim, Robert

I see your reasons.
I agree there aren't changes to do on _os.sync_

---

This is one of the implementation details to document, as in this proposal:
_https://github.com/orgs/micropython/discussions/11663_

At the moment, we have to update the official page:
_https://docs.micropython.org/en/latest/library/os.html_
with an addendum like

> On some ports this funciont isn't present because it isn't necessary to sync after writes to the file-system.

Who can do it?

Thanks for your attention.
Massimo

robert-hh, 2023-05-30:
> Who can do it?

Why not you?

massimosala, 2023-05-30:
> > Who can do it?
> 
> Why not you?

Hi Robert

I don't know how to.

Can you grant me edit access to the web pages?
Should I open a PR ?

Phew, sometimes it's more of a waste of time to fill out the forms than to do the job :-)

massimosala, 2023-05-30:
Sorry for the joke.
Is it enough to open a PR, specifying the web page and the text to add?

robert-hh, 2023-05-31:
As told in a different post, you have to make a PR, using the various git workflows. Either editing the file sin question online (easy), or by forking/cloning the repository, editing them locally and pushing a branch for being merged (more sophisticated). Since you will most likely have to compile the esp8266 firmware yourself, the second approach is reasonable for you.

massimosala, 2024-06-20:
Ho @robert-hh Robert

I am reviewing my old issues to close them.

For this one, I am interested in the quick approach

> either editing the file ... online

I go to the web page
[OS](https://docs.micropython.org/en/latest/library/os.html)
but I cannot find how to change it.

Pls help me.