micropython/micropython #11621: ESP32 I2S music nose

iBigDane, 2023-05-25:
[audio_pcm.zip](https://github.com/micropython/micropython/files/11565404/audio_pcm.zip)
I use version 1.19.1,esp32 with spiram,After I play a piece of music, there is always a noise after the music ends.After I modified the contents of line 465 of the machine_i2s.c  file to ```i2s_config.dma_buf_count = 3;```, the problem disappeared.I wonder why,thanks!

miketeachman, 2023-05-26:
There is a [bug in the ESP-IDF](https://github.com/espressif/esp-idf/issues/9882) which can cause this problem.  A possible workaround is to write a buffer with null samples (e.g. silence) immediately after the write with audible samples.

` audio_out.write(buffer_with_null_samples)`

You may need to experiment with different buffer sizes to eliminate the noise.

Another work around option is to experiment with different sizes of `ibuf`.  That is essentially what you are doing by modifying `i2s_config.dma_buf_count`.

jonnor, 2024-07-07:
There seems to be a fix/workaround available for this problem, so I think this issue can be closed.

closed 2024-07-07 by mattytrentini