micropython/micropython #11628: Error when importing const with leading underscore

stephanelsmith, 2023-05-25:
In short, if a variable is a micropython const with a leading underscore, getting an error on import.  Tried on esp32s3 as well as unix ports.

I've created a file defs.py with the following
```
from micropython import const
_A_CONST = const(10)
B_CONST = const(10)
_C_CONST = 10
```

OIn repl (or in another file), importing each of the options above:
```
>>> from defs import _A_CONST
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name _A_CONST
>>> from defs import B_CONST
>>> from defs import _C_CONST
```

dlech, 2023-05-25:
The leading underscore prevents the name from being added to the globals dict of the module. This is intended as an optimization.

If you want to take full advantage of `const()`, you need to redefine them (with leading underscore) in each file where they are used.

stephanelsmith, 2023-05-26:
Wow, fast response!  Roger roger.  Just a syntactical quirk I hadn't heard of yet, easy to work with.

Leading underscore aside, for taking full advantage of const(), do they need to be in the file itself or may they be imported.  I'm treating these a bit like #define directive in a .h file.

stephanelsmith, 2023-05-26:
Closing the issue, simple work-around.

closed 2023-05-26 by stephanelsmith