micropython/micropython #11629: When compiling and porting stm32H750 to micro python, an error is reported

klora-sun, 2023-05-26:
make BOARD=REG_H750 
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC system_stm32.c
In file included from ../../py/obj.h:32,
                 from ./pin.h:33,
                 from ./mphalport.h:3,
                 from ../../py/mphal.h:35,
                 from system_stm32.c:78:
../../py/obj.h: In function 'mp_obj_is_float':
../../py/misc.h:54:50: error: size of unnamed array is negative
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                  ^
../../py/obj.h:197:5: note: in expansion of macro 'MP_STATIC_ASSERT'
  197 |     MP_STATIC_ASSERT(sizeof(mp_float_t) <= sizeof(mp_obj_t));
      |     ^~~~~~~~~~~~~~~~
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [../../py/mkrules.mk:83：build-REG_H750/system_stm32.o] 错误 1

When compiling and porting stm32H750 to micro python, an error is reported，Please tell me what is wrong and how it should be changed

dpgeorge, 2023-05-26:
It looks like you have a custom board definition, and you have not configured it correctly.   Please share your board configuration files to make progress here with debugging.

klora-sun, 2023-05-26:
#define MICROPY_OBJ_REPR            (MICROPY_OBJ_REPR_C)
#define UINT_FMT                    "%u"
#define INT_FMT                     "%d"
typedef int mp_int_t;               // must be pointer size
typedef unsigned int mp_uint_t;     // must be pointer size

#define MICROPY_FATFS_EXFAT         (1)
#define MICROPY_HW_ENABLE_RTC       (1)
#define MICROPY_HW_ENABLE_RNG       (1)
#define MICROPY_HW_ENABLE_ADC       (1)
#define MICROPY_HW_ENABLE_DAC       (1)
#define MICROPY_HW_ENABLE_USB       (1)
#define MICROPY_HW_HAS_SWITCH       (1)
#define MICROPY_HW_HAS_FLASH        (1)
#define MICROPY_HW_ENABLE_SERVO     (1)
#define MICROPY_HW_ENABLE_TIMER     (1)
#define MICROPY_HW_ENABLE_SDCARD    (1)
#define MICROPY_HW_ENABLE_MMCARD    (0)

// Flash storage config
#define MICROPY_HW_SPIFLASH_ENABLE_CACHE            (1)
#define MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE    (0)

#define MICROPY_BOARD_STARTUP       REG_H7_board_startup
void REG_H7_board_startup(void);

#define MICROPY_BOARD_EARLY_INIT    REG_H7_board_early_init
void REG_H7_board_early_init(void);

#define MICROPY_BOARD_ENTER_BOOTLOADER(nargs, args) REG_H7_board_enter_bootloader()
void REG_H7_board_enter_bootloader(void);

void REG_H7_board_low_power(int mode);
#define MICROPY_BOARD_LEAVE_STOP    REG_H7_board_low_power(0);
#define MICROPY_BOARD_ENTER_STOP    REG_H7_board_low_power(1);
#define MICROPY_BOARD_ENTER_STANDBY REG_H7_board_low_power(2);

void REG_H7_board_osc_enable(int enable);
#define MICROPY_BOARD_OSC_ENABLE    REG_H7_board_osc_enable(1);
#define MICROPY_BOARD_OSC_DISABLE   REG_H7_board_osc_enable(0);

// PLL1 400MHz/50MHz for SDMMC and FDCAN
// USB and RNG are clocked from the HSI48
#define MICROPY_HW_CLK_PLLM             (5)
#define MICROPY_HW_CLK_PLLN             (160)
#define MICROPY_HW_CLK_PLLP             (2)
#define MICROPY_HW_CLK_PLLQ             (16)
#define MICROPY_HW_CLK_PLLR             (2)
#define MICROPY_HW_CLK_PLLVCI           (RCC_PLL1VCIRANGE_2)
#define MICROPY_HW_CLK_PLLVCO           (RCC_PLL1VCOWIDE)
#define MICROPY_HW_CLK_PLLFRAC          (0)

// PLL2 200MHz for FMC and QSPI.
#define MICROPY_HW_CLK_PLL2M            (5)
#define MICROPY_HW_CLK_PLL2N            (80)
#define MICROPY_HW_CLK_PLL2P            (2)
#define MICROPY_HW_CLK_PLL2Q            (2)
#define MICROPY_HW_CLK_PLL2R            (2)
#define MICROPY_HW_CLK_PLL2VCI          (RCC_PLL2VCIRANGE_2)
#define MICROPY_HW_CLK_PLL2VCO          (RCC_PLL2VCOWIDE)
#define MICROPY_HW_CLK_PLL2FRAC         (0)

// PLL3 160MHz for ADC and SPI123
#define MICROPY_HW_CLK_PLL3M            (5)
#define MICROPY_HW_CLK_PLL3N            (160)
#define MICROPY_HW_CLK_PLL3P            (2)
#define MICROPY_HW_CLK_PLL3Q            (5)
#define MICROPY_HW_CLK_PLL3R            (2)
#define MICROPY_HW_CLK_PLL3VCI          (RCC_PLL3VCIRANGE_2)
#define MICROPY_HW_CLK_PLL3VCO          (RCC_PLL3VCOWIDE)
[... 200 lines cut ...]
#define MICROPY_HW_ETH_MDC          (pin_C1)
#define MICROPY_HW_ETH_MDIO         (pin_A2)
#define MICROPY_HW_ETH_RMII_REF_CLK (pin_A1)
#define MICROPY_HW_ETH_RMII_CRS_DV  (pin_A7)
#define MICROPY_HW_ETH_RMII_RXD0    (pin_C4)
#define MICROPY_HW_ETH_RMII_RXD1    (pin_C5)
#define MICROPY_HW_ETH_RMII_TX_EN   (pin_G11)
#define MICROPY_HW_ETH_RMII_TXD0    (pin_G13)
#define MICROPY_HW_ETH_RMII_TXD1    (pin_G12)

#define MICROPY_HW_USB_VID                      0x2341
#define MICROPY_HW_USB_PID                      0x055B
#define MICROPY_HW_USB_PID_CDC_MSC              (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC_HID              (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC                  (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_MSC                  (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC2_MSC             (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC2                 (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC3                 (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC3_MSC             (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC_MSC_HID          (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC2_MSC_HID         (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC3_MSC_HID         (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_LANGID_STRING            0x409
Here is my config file

klora-sun, 2023-05-26:
> 
#define MICROPY_OBJ_REPR （MICROPY_OBJ_REPR_C）
#define UINT_FMT “%u” #define INT_FMT “%d”

typedef int mp_int_t;必须是指针大小
类型定义无符号整数mp_uint_t;必须是指针大小

#define MICROPY_FATFS_EXFAT （1） #define MICROPY_HW_ENABLE_RTC （1） #define MICROPY_HW_ENABLE_RNG （1） #define MICROPY_HW_ENABLE_ADC （1） #define MICROPY_HW_ENABLE_DAC （1） #define MICROPY_HW_ENABLE_USB （1） #define MICROPY_HW_HAS_SWITCH （1） #define MICROPY_HW_HAS_FLASH （1） #define MICROPY_HW_ENABLE_SERVO （1） #define MICROPY_HW_ENABLE_TIMER （1）

#define MICROPY_HW_ENABLE_SD卡 （1） #define MICROPY_HW_ENABLE_MMCARD （0）

闪存存储配置
#define MICROPY_HW_SPIFLASH_ENABLE_CACHE （1） #define MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE （0）

#define MICROPY_BOARD_STARTUP REG_H7_board_startup
空REG_H7_board_startup（空）;

#define MICROPY_BOARD_EARLY_INIT REG_H7_board_early_init
空REG_H7_board_early_init（空）;

#define MICROPY_BOARD_ENTER_BOOTLOADER（nargs， args） REG_H7_board_enter_bootloader（）
void REG_H7_board_enter_bootloader（void）;

无效REG_H7_board_low_power（整数模式）;
#define MICROPY_BOARD_LEAVE_STOP REG_H7_board_low_power（0）;
#define MICROPY_BOARD_ENTER_STOP REG_H7_board_low_power（1）;
#define MICROPY_BOARD_ENTER_STANDBY REG_H7_board_low_power（2）;

无效REG_H7_board_osc_enable（int enable）;
#define MICROPY_BOARD_OSC_ENABLE REG_H7_board_osc_enable（1）;
#define MICROPY_BOARD_OSC_DISABLE REG_H7_board_osc_enable（0）;

用于 SDMMC 和 FDCAN
的 PLL1 400MHz/50MHz // USB 和 RNG 从 HSI48
#define MICROPY_HW_CLK_PLLM （5） #define MICROPY_HW_CLK_PLLN （160） #define MICROPY_HW_CLK_PLLP （2） #define MICROPY_HW_CLK_PLLQ （16） #define MICROPY_HW_CLK_PLLR （2） #define MICROPY_HW_CLK_PLLVCI （RCC_PLL1VCIRANGE_2） #define MICROPY_HW_CLK_PLLVCO （RCC_PLL1VCOWIDE） #define MICROPY_HW_CLK_PLLFRAC （0）

PLL2 200MHz，用于FMC和QSPI。
#define MICROPY_HW_CLK_PLL2M （5） #define MICROPY_HW_CLK_PLL2N （80） #define MICROPY_HW_CLK_PLL2P （2） #define MICROPY_HW_CLK_PLL2Q （2） #define MICROPY_HW_CLK_PLL2R （2） #define MICROPY_HW_CLK_PLL2VCI （RCC_PLL2VCIRANGE_2） #define MICROPY_HW_CLK_PLL2VCO （RCC_PLL2VCOWIDE） #define MICROPY_HW_CLK_PLL2FRAC （0）

用于 ADC 和 SPI3 的 PLL160 123MHz
#define MICROPY_HW_CLK_PLL3M （5） #define MICROPY_HW_CLK_PLL3N （160） #define MICROPY_HW_CLK_PLL3P （2） #define MICROPY_HW_CLK_PLL3Q （5） #define MICROPY_HW_CLK_PLL3R （2） #define MICROPY_HW_CLK_PLL3VCI （RCC_PLL3VCIRANGE_2） #define MICROPY_HW_CLK_PLL3VCO （RCC_PLL3VCOWIDE） #define MICROPY_HW_CLK_PLL3FRAC （0）

旁路模式下的 HSE。
#define MICROPY_HW_CLK_USE_BYPASS （0）

总线时钟分频器值
#define MICROPY_HW_CLK_AHB_DIV （RCC_HCLK_DIV2） #define MICROPY_HW_CLK_APB1_DIV （RCC_APB1_DIV2） #define MICROPY_HW_CLK_APB2_DIV （RCC_APB2_DIV2） #define MICROPY_HW_CLK_APB3_DIV （RCC_APB3_DIV2） #define MICROPY_HW_CLK_APB4_DIV （RCC_APB4_DIV2）

外设时钟源
#define MICROPY_HW_RCC_HSI48_STATE （RCC_HSI48_ON） #define MICROPY_HW_RCC_USB_CLKSOURCE （RCC_USBCLKSOURCE_HSI48） #define MICROPY_HW_RCC_RTC_CLKSOURCE （RCC_RTCCLKSOURCE_LSI） #define MICROPY_HW_RCC_FMC_CLKSOURCE （RCC_FMCCLKSOURCE_PLL2） #define MICROPY_HW_RCC_RNG_CLKSOURCE （RCC_RNGCLKSOURCE_HSI48） #define MICROPY_HW_RCC_ADC_CLKSOURCE （RCC_ADCCLKSOURCE_PLL3）

#define MICROPY_HW_RCC_SDMMC_CLKSOURCE （RCC_SDMMCCLKSOURCE_PLL） #define MICROPY_HW_RCC_FDCAN_CLKSOURCE （RCC_FDCANCLKSOURCE_PLL） #define MICROPY_HW_RCC_SPI123_CLKSOURCE （RCC_SPI123CLKSOURCE_PLL3） #define MICROPY_HW_RCC_I2C123_CLKSOURCE （RCC_I2C123CLKSOURCE_D2PCLK1） #define MICROPY_HW_RCC_QSPI_CLKSOURCE （RCC_QSPICLKSOURCE_PLL2）

有一个外部 32kHz 振荡器
#define RTC_ASYNCH_PREDIV （0） #define RTC_SYNCH_PREDIV （0x7fff） #define MICROPY_HW_RTC_USE_BYPASS （0） #define MICROPY_HW_RTC_USE_US （1） #define MICROPY_HW_RTC_USE_CALOUT （0）

#if （MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE == 0） // QSPI 闪存 https://github.com/micropython/micropython/issues/1 用于存储
#define MICROPY_HW_QSPI_PRESCALER （2） // 100MHz
#define MICROPY_HW_QSPIFLASH_SIZE_BITS_LOG2 （27）
#define MICROPY_HW_SPIFLASH_SIZE_BITS （128 * 1024 * 1024） #define MICROPY_HW_QSPIFLASH_CS （pyb_pin_QSPI2_CS） #define MICROPY_HW_QSPIFLASH_SCK （pyb_pin_QSPI2_CLK） #define MICROPY_HW_QSPIFLASH_IO0 （pyb_pin_QSPI2_D0）

#define MICROPY_HW_QSPIFLASH_IO1 （pyb_pin_QSPI2_D1） #define MICROPY_HW_QSPIFLASH_IO2 （pyb_pin_QSPI2_D2） #define MICROPY_HW_QSPIFLASH_IO3 （pyb_pin_QSPI2_D3）

SPI 闪存 https://github.com/micropython/micropython/issues/1，块设备配置
外部常量结构_mp_spiflash_config_t spiflash_config;
外部结构_spi_bdev_t spi_bdev;
#define MICROPY_HW_BDEV_IOCTL（op， arg） （
（op） == BDEV_IOCTL_NUM_BLOCKS ？（MICROPY_HW_SPIFLASH_SIZE_BITS / 8 / FLASH_BLOCK_SIZE） ：
（op） == BDEV_IOCTL_INIT ？spi_bdev_ioctl（&spi_bdev， （op）， （uint32_t）&spiflash_config） ：
spi_bdev_ioctl（&spi_bdev， （op）， （arg）） ）
[... 76 lines cut ...]
// 毫秒

SDRAM 配置
#define MICROPY_HW_FMC_SDCKE0 （pin_C3） #define MICROPY_HW_FMC_SDNE0 （pin_C2） #define MICROPY_HW_FMC_SDNBL0 （pin_E0） #define MICROPY_HW_FMC_SDNBL1 （pin_E1） #define MICROPY_HW_FMC_SDCLK （pin_G8） #define MICROPY_HW_FMC_SDNCAS （pin_G15） #define MICROPY_HW_FMC_SDNRAS （pin_F11） #define MICROPY_HW_FMC_SDNWE （pin_C0） #define MICROPY_HW_FMC_BA0 （pin_G4）

#define MICROPY_HW_FMC_BA1 （pin_G5） #define MICROPY_HW_FMC_NBL0 （pin_E0） #define MICROPY_HW_FMC_NBL1 （pin_E1） #define MICROPY_HW_FMC_A0 （pin_F0） #define MICROPY_HW_FMC_A1 （pin_F1） #define MICROPY_HW_FMC_A2 （pin_F2） #define MICROPY_HW_FMC_A3 （pin_F3） #define MICROPY_HW_FMC_A4 （pin_F4） #define MICROPY_HW_FMC_A5 （pin_F5） #define MICROPY_HW_FMC_A6 （pin_F12）

#define MICROPY_HW_FMC_A7 （pin_F13） #define MICROPY_HW_FMC_A8 （pin_F14） #define MICROPY_HW_FMC_A9 （pin_F15） #define MICROPY_HW_FMC_A10 （pin_G0） #define MICROPY_HW_FMC_A11 （pin_G1） #define MICROPY_HW_FMC_A12 （pin_G2） #define MICROPY [... cut ...]  MICROPY_HW_USB_PID_MSC （MICROPY_HW_USB_PID） #define MICROPY_HW_USB_PID_CDC2_MSC （MICROPY_HW_USB_PID） #define MICROPY_HW_USB_PID_CDC2 （MICROPY_HW_USB_PID）

#define MICROPY_HW_USB_PID_CDC3 （MICROPY_HW_USB_PID） #define MICROPY_HW_USB_PID_CDC3_MSC （MICROPY_HW_USB_PID） #define MICROPY_HW_USB_PID_CDC_MSC_HID （MICROPY_HW_USB_PID） #define MICROPY_HW_USB_PID_CDC2_MSC_HID （MICROPY_HW_USB_PID） #define MICROPY_HW_USB_PID_CDC3_MSC_HID （MICROPY_HW_USB_PID）

#define MICROPY_HW_USB_LANGID_STRING 0x409
这是我的配置文件

klora-sun, 2023-05-26:
> It looks like you have a custom board definition, and you have not configured it correctly. Please share your board configuration files to make progress here with debugging.

#define MICROPY_OBJ_REPR            (MICROPY_OBJ_REPR_C)
#define UINT_FMT                    "%u"
#define INT_FMT                     "%d"
typedef int mp_int_t;               // must be pointer size
typedef unsigned int mp_uint_t;     // must be pointer size

#define MICROPY_FATFS_EXFAT         (1)
#define MICROPY_HW_ENABLE_RTC       (1)
#define MICROPY_HW_ENABLE_RNG       (1)
#define MICROPY_HW_ENABLE_ADC       (1)
#define MICROPY_HW_ENABLE_DAC       (1)
#define MICROPY_HW_ENABLE_USB       (1)
#define MICROPY_HW_HAS_SWITCH       (1)
#define MICROPY_HW_HAS_FLASH        (1)
#define MICROPY_HW_ENABLE_SERVO     (1)
#define MICROPY_HW_ENABLE_TIMER     (1)
#define MICROPY_HW_ENABLE_SDCARD    (1)
#define MICROPY_HW_ENABLE_MMCARD    (0)

// Flash storage config
#define MICROPY_HW_SPIFLASH_ENABLE_CACHE            (1)
#define MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE    (0)

#define MICROPY_BOARD_STARTUP       REG_H7_board_startup
void REG_H7_board_startup(void);

#define MICROPY_BOARD_EARLY_INIT    REG_H7_board_early_init
void REG_H7_board_early_init(void);

#define MICROPY_BOARD_ENTER_BOOTLOADER(nargs, args) REG_H7_board_enter_bootloader()
void REG_H7_board_enter_bootloader(void);

void REG_H7_board_low_power(int mode);
#define MICROPY_BOARD_LEAVE_STOP    REG_H7_board_low_power(0);
#define MICROPY_BOARD_ENTER_STOP    REG_H7_board_low_power(1);
#define MICROPY_BOARD_ENTER_STANDBY REG_H7_board_low_power(2);

void REG_H7_board_osc_enable(int enable);
#define MICROPY_BOARD_OSC_ENABLE    REG_H7_board_osc_enable(1);
#define MICROPY_BOARD_OSC_DISABLE   REG_H7_board_osc_enable(0);

// PLL1 400MHz/50MHz for SDMMC and FDCAN
// USB and RNG are clocked from the HSI48
#define MICROPY_HW_CLK_PLLM             (5)
#define MICROPY_HW_CLK_PLLN             (160)
#define MICROPY_HW_CLK_PLLP             (2)
#define MICROPY_HW_CLK_PLLQ             (16)
#define MICROPY_HW_CLK_PLLR             (2)
#define MICROPY_HW_CLK_PLLVCI           (RCC_PLL1VCIRANGE_2)
#define MICROPY_HW_CLK_PLLVCO           (RCC_PLL1VCOWIDE)
#define MICROPY_HW_CLK_PLLFRAC          (0)

// PLL2 200MHz for FMC and QSPI.
#define MICROPY_HW_CLK_PLL2M            (5)
#define MICROPY_HW_CLK_PLL2N            (80)
#define MICROPY_HW_CLK_PLL2P            (2)
#define MICROPY_HW_CLK_PLL2Q            (2)
#define MICROPY_HW_CLK_PLL2R            (2)
#define MICROPY_HW_CLK_PLL2VCI          (RCC_PLL2VCIRANGE_2)
#define MICROPY_HW_CLK_PLL2VCO          (RCC_PLL2VCOWIDE)
#define MICROPY_HW_CLK_PLL2FRAC         (0)

// PLL3 160MHz for ADC and SPI123
#define MICROPY_HW_CLK_PLL3M            (5)
#define MICROPY_HW_CLK_PLL3N            (160)
#define MICROPY_HW_CLK_PLL3P            (2)
#define MICROPY_HW_CLK_PLL3Q            (5)
#define MICROPY_HW_CLK_PLL3R            (2)
[... 202 lines cut ...]
#define MICROPY_HW_ETH_MDC          (pin_C1)
#define MICROPY_HW_ETH_MDIO         (pin_A2)
#define MICROPY_HW_ETH_RMII_REF_CLK (pin_A1)
#define MICROPY_HW_ETH_RMII_CRS_DV  (pin_A7)
#define MICROPY_HW_ETH_RMII_RXD0    (pin_C4)
#define MICROPY_HW_ETH_RMII_RXD1    (pin_C5)
#define MICROPY_HW_ETH_RMII_TX_EN   (pin_G11)
#define MICROPY_HW_ETH_RMII_TXD0    (pin_G13)
#define MICROPY_HW_ETH_RMII_TXD1    (pin_G12)

#define MICROPY_HW_USB_VID                      0x2341
#define MICROPY_HW_USB_PID                      0x055B
#define MICROPY_HW_USB_PID_CDC_MSC              (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC_HID              (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC                  (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_MSC                  (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC2_MSC             (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC2                 (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC3                 (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC3_MSC             (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC_MSC_HID          (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC2_MSC_HID         (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_PID_CDC3_MSC_HID         (MICROPY_HW_USB_PID)
#define MICROPY_HW_USB_LANGID_STRING            0x409
Here is my config file

dpgeorge, 2023-05-26:
> #define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_C)

You can't use this object representation with double precision float (which the H750 has).

I suggest just removing this line and using the default object representation A.  (You'll also need to remove the 4 lines after it as well.)

klora-sun, 2023-05-26:
> > #define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_C)
> 
> You can't use this object representation with double precision float (which the H750 has).
> 
> I suggest just removing this line and using the default object representation A. (You'll also need to remove the 4 lines after it as well.)

OK, let me try

klora-sun, 2023-05-26:
> > #define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_C)
> 
> You can't use this object representation with double precision float (which the H750 has).
> 
> I suggest just removing this line and using the default object representation A. (You'll also need to remove the 4 lines after it as well.)

Hello, after deletion, it does work properly. But I would like to ask this object representation with double precision float (which the H750 has).What does it mean, and how can I avoid similar problems the next time I want to continue compiling the firmware. Will there be a description of the questions in the data sheet of the chip?

klora-sun, 2024-08-26:
> Hello klora-sun, would you mind, if i ask you for your port for stm32h750 board?

sorry, I’m too busy in this month. I would like to help you to solve some problems,but I don’t have too much time .But , if you have some ideas,we can discuss.thanks for your inviteing.

获取 Outlook for iOS<https://aka.ms/o0ukef>

jonnor, 2024-09-01:
The problem looks to have been resolved, will close this issue.
Information about the MicroPython object model can be found in https://docs.micropython.org/en/latest/develop/memorymgt.html#the-object-model and in `py/mpconfig.h` in the codebase.

closed 2024-09-01 by jonnor