micropython/micropython #11635: I2S MONO format does not work

wuhao19920619, 2023-05-27:
I am running MicroPython v1.20.0 on 2023-04-26; ESP32S3 module (spiram octal) with ESP32S3

Here is my code:

```
from machine import Pin, I2S
import array

sck_pin = Pin(14)   # Serial clock output
ws_pin = Pin(13)    # Word clock output
sd_pin = Pin(12)    # Serial data output

audio = I2S(0,
            sck=sck_pin, ws=ws_pin, sd=sd_pin,
            mode=I2S.TX,
            bits=16,
            format=I2S.MONO,
            rate=8000,
            ibuf=20000)

d = array.array("H", [0]*2)
d[0] = 0b1111111111111111
d[1] = 0b1000100010001000
print(d)
while True:
    audio.write(d)
```

According to the I2S protocol, 16 bits data should be transfered to both left channel and right channel. Therefore d[0] should be transfered twice in a cycle of Word clock, and also the d[1].

However, This is what I get from oscilloscope:
[image]

The yellow line is the ws_pin and the green line is sd_pin.

Obviousely it transfer the d[0] to left channel and d[1] to right channel. This is the Stereo format.

Also, May I ask what is the first parameter `id` of `I2S()` stand for? The document is quite unclear. Something weird with this parameter.

If I set it to 0 or 1, there is nothing different. I can't set it to 2.

However, If you put this code in the beginning:
```
from machine import Pin, I2S
import array

sck_pin = Pin(14)   # Serial clock output
ws_pin = Pin(13)    # Word clock output
sd_pin = Pin(12)    # Serial data output

audio = I2S(1,
            sck=sck_pin, ws=ws_pin, sd=sd_pin,
            mode=I2S.TX,
            bits=16,
            format=I2S.MONO,
            rate=8000,
            ibuf=20000)

aduio.deinit()

audio = I2S(0,
            sck=sck_pin, ws=ws_pin, sd=sd_pin,
            mode=I2S.TX,
            bits=16,
            format=I2S.MONO,
            rate=8000,
            ibuf=20000)
```
you will surprisingly find on the oscilloscope that the i2s protocol turn from the general `Phillips Standard` to the `left-justified`. This only occur when you define the object twice with different `id` (of course deinit it in between)

ma261065, 2023-05-29:
ESP32 contains two I2S peripherals. The first parameter specifies which of the two internal I2S peripherals you are working with.
See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

There is definitely something weird going on with I2S on the S3 chip. See my issue at https://github.com/micropython/micropython/issues/11245

Even though it seems to be a different issue, maybe you could try my fix and see if the misbehaving stops.

wuhao19920619, 2023-06-04:
Well I don't understand but if I put these codes in front of audio object, everything will be okay:
```
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
if wlan.isconnected():
    wlan.disconnect()
wlan.connect('ICMA', 'icmaicma')
```
I tested your solution at #11245 , it works good even without the code above.

There is also some bugs in soft reboot. Like the codes above, if I remove the `if wlan.isconnected():` it will raise an error.

Also I tried to modify the `sck_pin` `ws_pin` `sd_pin` without the hard reset in between. During the new code running, both the old pin and new pin have the same signal output.

dpgeorge, 2024-03-08:
This may be fixed by 0b145fdc2d4857a77f9f0a3662646d52da61d5b2