micropython/micropython #11636: network.WLAN(network.STA_IF) will create a access point if connection failed.

wuhao19920619, 2023-05-28:
I am running MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266

I put the following code in boot.py and nothing in main.py:
```
from debugfun import breakpoint, vars
print("Activating network")
import network
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect('ssid', 'password')
```
if my router power off when esp8266 connected, Then the esp8266 will create a new access point with the same ssid and password as my router. It's DHCP is `192168.4.*`.

It will prevent the other devices to reconnect to the router.

glenn20, 2023-06-05:
I have not tested this bug myself yet, but one thing to check: 

- the ESP8266 automatically saves the wifi configuration from previous boots in nvs storage and restarts with the same settings. So, if you have ever configured the AP in the past while experimenting or accidentally, it *may* be restarting the AP automatically. I regard this as a mis-feature of the esp8266.

On the ESP8266 it is safest to explicitly set both interfaces to the mode you want on every boot: ie. `network.WLAN(network.AP_IF).active(False)`.

I'm hesitant to attribute your issue to this cause without further testing as it is not quite the behaviour I would expect if this was the cause, but thought it worth sharing.

glenn20, 2023-06-05:
Oh - and one more thing. Make sure you are power cycling or hard resetting the esp8266 as soft resets don't re-initialise the wifi peripheral.

glenn20, 2023-06-06:
> I don't know if this mis-feature comes from esp-idf or micropython. Can it be fixed by micropython?

It is a default behaviour of the espressif SDK on the esp8266 (esp-idf is for the esp32). It can be overridden, and I have submitted a PR (#9460) to micropython to allow users to turn it off, but it has not been merged with micropython main.

glenn20, 2023-06-06:
> So does it mean that the wlan object that I create is working on STA_IF mode. While at the same time there is also another wlan object working in the AP mode? 

> There is two object working at the same time with differect mode but on the same wifi peripheral?

Yes. There is only one radio on the esp8266 (or esp32) device, but you can connect your STA_IF to an access point while also running an access point on the AP_IF interface. They must operate on the same wifi channel, as there is only one radio device.