micropython/micropython #11638: Type Inference issue when using struct in viper

GuillaumeLeclerc, 2023-05-28:
```
MY_TYPE = {
    "var": 0 | uctypes.UINT32
}
storage = bytearray(uctypes.sizeof(MY_TYPE))

@micropython.viper
def bugreport():
    data = uctypes.struct(ptr32(storage), MY_TYPE)
    data.var += int(1)
```

For some reason data.var is an object even though it's type is clearly defined from the struct. the following error occurs:

```
ViperTypeError: can\'t do binary op between \'object\' and \'int\'
```

robert-hh, 2023-05-28:
It looks like you found way to do it right in the #11639 example. viper's scope is pure local. It does not reflect the uctypes declaration.

GuillaumeLeclerc, 2023-05-28:
Any technical reason why it's not used at compilation to compute offsets ? Could it be considered a feature proposal ?