micropython/micropython #11639: Hidden Extraneous memory allocation

GuillaumeLeclerc, 2023-05-28:
```python
import uctypes
import gc
import micropython

MY_TYPE = {
    'var': 0 | uctypes.UINT32,
    "previous_measurements": (8 | uctypes.ARRAY, 5 | uctypes.UINT32),
}
buffer = bytearray(uctypes.sizeof(MY_TYPE))

@micropython.viper
def bugreport():
    while True:
        data = uctypes.struct(ptr32(buffer), MY_TYPE)
        data.var = int(data.var) + 1

if __name__ == '__main__':
    gc.collect()
    gc.disable()
    bugreport()
```

This code crash but shouldn't ever allocate memory after the garbage collector is disabled. 

When the `bugreport` is compiled, offsets should be computable by reading `MY_TYPE`, and memory access replaced accordingly. 

I thought that allocating all the data ahead of time in a struct would help make my code jitter free but it seems impossible in micropython.

robert-hh, 2023-05-28:
You could write the inner loop as:
```
@micropython.viper
def bugreport():
    while True:
        data = ptr32(buffer)
        data[0] += 1
```

GuillaumeLeclerc, 2023-05-28:
But it takes away all the maintainability of the code. Instead of having attribute names, now everything is an index.

peterhinch, 2023-05-28:
> shouldn't ever allocate memory after the garbage collector is disabled.

Disabling GC does not prevent allocation. It simply prevents garbage collection.

jimmo, 2023-05-29:
Unfortunately there isn't much we can do about this as `uctypes.struct` must allocate an object.

> But it takes away all the maintainability of the code. Instead of having attribute names, now everything is an index.

If it helps you can use `const()` to at least make the indicies readable.

```
_OFFSET_OF_FOO = const(0)

@micropython.viper
def bugreport():
    while True:
        data = ptr32(buffer)
        data[_OFFSET_OF_FOO] += 1
```

FYI, a good way to check for extraneous allocations in specific bits of code is to explicitly lock the heap (rather than waiting for the GC to fail to collect).

```
try:
  micropython.heap_lock()
  ... code to check ...
finally:
  micropython.heap_unlock()
```

KubaO, 2023-06-27:
A workaround is to pre-allocate the accessor, instead of creating them on-the-fly:

```
import uctypes
import gc
import micropython

MY_TYPE = {
    'var': 0 | uctypes.UINT32,
    "previous_measurements":  4 | uctypes.UINT32,
}
buffer = bytearray(uctypes.sizeof(MY_TYPE))

@micropython.viper
def make_data():
    data = uctypes.struct(ptr32(buffer), MY_TYPE)
    return data

data = make_data()

@micropython.viper
def good():
    global data
    micropython.heap_lock()
    data.var = int(data.var) + 1
    micropython.heap_unlock()
    print("good succeeded")

@micropython.viper
def bad():
    micropython.heap_lock()
    data = uctypes.struct(ptr32(buffer), MY_TYPE)
    data.var = int(data.var) + 1
    micropython.heap_unlock()
    print("bad succeeded")

if __name__ == '__main__':
    print("hello")
    gc.collect()

    try:
        bad()
    except MemoryError:
        micropython.heap_unlock()
        print("bad failed")

    try:
        good()
    except MemoryError:
        micropython.heap_unlock()
        print("good failed")

    print("bye")

```

Output:
```
hello
bad failed
good succeeded
bye
```

jonnor, 2024-07-13:
As mentioned the uctypes.struct() must allocate an object. And there seems to be go tips here on how to reduce/remove allocations by changing the code. Proposing to close this issue.

closed 2024-07-13 by peterhinch