micropython/micropython #11641: Reading from a PIO queue allocates memory in viper mode

GuillaumeLeclerc, 2023-05-28:
```python
import gc
from rp2 import StateMachine, asm_pio
import micropython

micropython.alloc_emergency_exception_buf(100)

@asm_pio()
def test():
    wrap_target()
    mov(isr, x)
    push()
    wrap()

@micropython.viper
def bugreport():
    while True:
        m = StateMachine(0)
        m.get()

if __name__ == '__main__':
    machine = StateMachine(0, test, freq=-1)
    machine.active(True)
    gc.disable()
    bugreport()
```

This code fail even though according to the datasheet, reading from the PIO RX queue should be just a register read. Here it tries to allocate memory and fails.

mendenm, 2023-09-30:
Aren't you allocating a new StateMachine() object  inside the loop, and with gc turned off, running out of memory with abandoned StateMachine objects?  If you do:

```python
@micropython.viper
def bugreport():
    m = machine #pick up global StateMachine object created in __main__
    while True:
        m.get()
```

does it work?

GuillaumeLeclerc, 2023-10-01:
That's a good point. However I moved on to C++ as it was too hard to hit any timings with the garbage collector and I don't have any board programmed with Micropython anymore...

I guess I can close this issue. Thanks for answering!

closed 2023-10-01 by GuillaumeLeclerc