micropython/micropython #11656: Flashing micropython does not work with ESP-01

wohltat, 2023-05-29:
I cannot successfully flash micropython from the thonny gui anymore. It worked in previous versions.
It uploads without errors but when i try to run the TX or RX indication LED blinks rapidly without stopping. 
When i try to connect i just see gibberish like the following on the console:

``` 
��nn'l`�nn$`nr���nl`r��nl`��p�nn�llllb'|l`rl�l�b|���rblb�nnlnn�bbp�$blrlp�n��lbn�n��$�b��nn'l�l`�nn$`nr���n�`p�n�r�bn�nll�b��nn'l`�nn$`nr���n��l`p�n�r�bn�nlb��nn'l`�nn$`nr���nl`r��nl`��p�nn�llllb'|l`rl�l�b|
```
[image]

I manually flashed the v1.20 micropython on an ESP-01 (ESP8266) and everything works fine. No unstoppable LED blinking and i can connect to the board with thonny.
`> build/thonny/thonny_venv/bin/esptool.py --port /dev/ttyUSB* --baud 460800 write_flash --flash_size=detect 0 Downloads/esp8266-1m-20230426-v1.20.0.bin`

I had a closer look on the upload log in thonny and i copied the command line. The culprit seems to be `--flash_size keep`. If i change it to `--flash_size detect` it works again. Everything in the output looks identical except for the line `Auto-detected Flash size: 1MB`.

What about adding that to the list of `Flash mode` or even make it the default?

-----------

version: Thonny 4.1.0b1

robert-hh, 2023-05-29:
You should report that at the Thonny repository https://github.com/thonny/thonny. Issues with Thonny are not MicroPython issues.

wohltat, 2023-05-29:
Sorry, actually i intended to do that  i must have mixed up my browser tabs.
We can delete this thread then.

closed 2023-05-29 by wohltat

niftyhacking, 2023-05-30:
I see that it was closed as a thonny issue (OK).

Two cents: ���nr���n ��l`� and friends are classic baud rate (tty param)
mismatches.
I see for the most part  DEFAULT_BAUDRATE   115200  in source.
The first message from the first poster contained --baud 460800  so I
wonder.
Often limits on data transfer rates can help with data errors.
I quickly searched for more ways to limit and control and did not find any
on the micropython side.
Is there a guide for standard data rates and params and if not standard a
board should document their enhancements.

Now off to look at thonny command line options.

wohltat, 2023-05-30:
on 115200 there is gibberish as well. No idea why this happens.

But this is kind of solved for me now:
https://github.com/thonny/thonny/issues/2801

robert-hh, 2023-05-30:
When the ESP8266 starts, the first boot message is sent at a baud rate of 74880. Decoding that at 115200 baud results in gibberish.

wohltat, 2023-05-30:
when i connect with screen at 74880 baud it also looks like garbage.

jouellnyc, 2024-06-10:
FYI - I added to https://github.com/thonny/thonny/issues/2801 - in case anyone has a similar issue flash w/the esp01s.