micropython/micropython #11666: esp8266 : available heap: DRAM ?

massimosala, 2023-05-30:
MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266
MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266
MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266
Reading this doc
https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map
there is a region of memory, called DRAM: 

> User data RAM. Available to applications.

It is 80 KB.

Does the micropython firmware use it?
If not, is it possible to merge this region the the heap?

---

MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266

At startup I have about 35 KB of free heap, running only this _boot.py_

```
from micropython import alloc_emergency_exception_buf
alloc_emergency_exception_buf(100)		# give traceback info when RAM allocation would fail

import gc
gc.enable()
#gc.threshold((gc.mem_free() + gc.mem_alloc()) // 4)
gc.threshold(gc.mem_free() // 4 + gc.mem_alloc())

import esp
esp.osdebug(None)
esp.sleep_type(esp.SLEEP_NONE)

from machine import freq
freq(160000000)				# CPU at 160 MHz

try :
	from sys import tracebacklimit
	tracebacklimit(10)
except :
	pass

#import uos
# detach REPL UART (GPIO1, 3; disable print, also via USB)
#uos.dupterm(None, 1) # disable REPL on UART(0)

gc.collect()
print("boot.py", gc.mem_free())
```

jimmo, 2023-05-31:
This has been discussed elsewhere but the simple answer is yes, the heap lives in "data RAM" (which is actually SRAM, "DRAM" is something else). However the heap needs a contiguous range of memory, and so we use the largest possible contiguous range left over from the ESP8266's RTOS.

You also have to leave room for the C stack, as well as global interpreter state, and also state used for the WiFi driver. The summary is that of the 80kiB of SRAM, 38kiB is available for the MicroPython heap.

Your boot.py does already allocate some things, but also there's code that runs before boot.py to initialise the filesystem etc, so that accounts for the missing 3kiB to get your result of 35kiB. You can use `micropython.mem_info(1)` to investigate further.

closed 2023-05-31 by jimmo

robert-hh, 2023-05-31:
@massimosala Since the heap size of the ESP8266 is very limited, and thus the size of the code you can run from RAM. The only workaround is putting the code into flash. At the moment, this is available by "freezing" them during the firmware build process. So you have to set up a build environment. That's a little bi tedious, but worth the effort. Once it runs, it's stable.
As another option you could change the chip model and use e.g. an esp32, which offers more RAM, especially with the SPIRAM variants, and more interface options.