micropython/micropython #11667: viper: uint: support div/mod; support specific data types

massimosala, 2023-05-30:
General topic, not related to a specific micro.

Viper is very goo to optimize functions, like custom hashes, performing only calculation on integer.

1) First problem: div/mod

These functions work with uint.
Some like the adler_32 checksum require the modulus operation...

> ViperTypeError: div/mod not implemented for uint

2) Code optimization

Viper provides two generics int / uint.
Is it possible to have specialized types ?

> int8, int16, int32
> uint8, uint16, uint32
 
Thus the compiler can further optimize cpu register allocations and hopefully issue specialized instructions for the int width of the operands.

I've seen in other (non-micropython) environments that using specialized types leads to a sensible speed gain.

jonnor, 2024-09-02:
What is some example code that triggers this exception?

agatti, 2024-09-04:
In theory this can be fixed by adding new functions to native emitters (as in `ASM_DIV_REG_REG` and `ASM_DIVU_REG_REG`) and add support for that in `emitnative.c`.  However, `armv6` doesn't have any division opcode at all¹, so you'd need to extend the native code interface in `mp_fun_table_t` with a variant of `small_int_floor_divide` (and don't forget to update the natmod tooling!).

Quite a bit of work when you can perform the signed<>unsigned conversion yourself beforehand :)

¹ need to check about Xtensa if it has division opcodes, but the point still stands.