micropython/micropython #11678: RP2 UART timeout is incorrect

peterhinch, 2023-06-01:
To repro link pins 0 and 1 and paste the following:
```python
from machine import UART, Pin
uart = UART(0, 115200, rx=Pin(1, Pin.IN), tx=Pin(0, Pin.OUT), timeout=1000_000, timeout_char=1000_000)
uart.readline()
```
It times out after 15s rather than 16.6minutes.

Setting a long timeout is a hack to resolve https://github.com/micropython/micropython/issues/10867. In my opinion it should be possible to disable the timeout (set it to infinity). In `uasyncio` systems, `uasyncio` should provide any timeouts required by the application.

robert-hh, 2023-06-01:
I will look for it. I am never sure whether the timeout is given in ms or character times.

robert-hh, 2023-06-01:
The reason is simple. The element for storing the timeout are defined as `uint16_t`. So 1000_000 maps to 0x4240 or 16960.

peterhinch, 2023-06-01:
OK, but there is a real problem here. In my testing when the UART RX times out, it returns `b'\xff`. It is not possible for `uasyncio.StreamReader` to distinguish the timeout case from the case where the target sends that single character. If the UART returned `None` it would be possible in principle for `StreamReader` to cope, but in my view this would be a hack as the problem is in the UART not in `StreamReader`.

Would it be possible to have `timeout=-1, timeout_char=-1` disable timeouts altogether? This would fix the `uasyncio` issue.

robert-hh, 2023-06-01:
Re-doing a test: 
Calling uart_read(10) or uart.read_line() with timeout the first time after instantiating the UART, it returns b'\x00'. On further calls, it returns `None`.

robert-hh, 2023-06-01:
I wonder if there is really a need to increase the timeout range or if should just be documented. It's a simple change. The biggest effort is testing. @dpgeorge, what do you suggest?

robert-hh, 2023-06-01:
> Would it be possible to have timeout=-1, timeout_char=-1 disable timeouts altogether? This would fix the uasyncio issue.

At the moment, the default value of the keyword argument is -1, telling that timeout is set to a default start value, usually 0 or 1. The default value could as well be `None`, making it possible to distinguish between 'no value', -1 and >= 0. But I would not like to add additional tests into the wait loops. Instead, -1 could set the timeout to the maximum value, which with a 32 bit data type is about 50 days. But such a long timeout could by set without that change.

peterhinch, 2023-06-01:
> I wonder if there is really a need to increase the timeout range

Increasing the range doesn't fix the problem. How big should it be? There seem to be two approaches:
 1. Have a "no timeout" option.
 2. Keep the UART drivers as they are, ensuring that a timeout always returns `None`. Then have `StreamReader` discard `None` responses.

A mandatory timeout on a UART is odd (IMO), but option 2 may be easiest to implement. But this does need fixing: it keeps cropping up. @dpgeorge, @jimmo what are your views?

robert-hh, 2023-06-01:
I used the little script below to test the 32 bit timeout. On all tested board uart.read_line() (and uart.read(n)) returned `None` on timeout. Tested with RP2, STM32, ESP32, ESP8266, MIMXRT, nrf, SAMD, Renesas-RA. The while loop after instantiation of the UART is need to catch a false byte which sometimes is seen when the pins settle.  What was the situation in which uart.read() returned 0xff.

```
from machine import UART, Pin
import time
timeout = 100_000
uart = UART(0, 115200, rx=Pin(1, Pin.IN), tx=Pin(0, Pin.OUT), timeout=timeout, timeout_char=timeout)
time.sleep_ms(1)
while uart.any():
    uart.read(1)

start = time.time()
b = uart.readline()
print(b)
print("Time expired", time.time() - start)
```