micropython/micropython #11682: Busy `try: except:` loop prevents `KeyboardInterrupt` to break the loop

josuah, 2023-06-01:
When inserting this in paste mode:

```python
while True:
    try:
        for i in range(1000):
            pass
        raise(TypeError("Any error you actually want to catch"))
    except:
        pass
```

Hitting Ctrl-C (KeyboardInterrupt) does not trigger a KeyboardInterrupt anymore.
We stay in the end of paste mode, the loop being executed continuously.

On the other hand, when reducing 1000 to 100, we end-up seeing a KeyboardInterrupt:

```python
MicroPython v1.20.0-117-ged7a3b11d on 2023-06-01; Monocle on nRF52832
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== while True:
===     try:
===         for i in range(100):
===             pass
===         raise(TypeError("Any error you actually want to catch"))
===     except:
===         pass
=== 
Traceback (most recent call last):
  File "<stdin>", in <module>
KeyboardInterrupt: 
```

The bug was reproduced with https://github.com/brilliantlabsAR/monocle-micropython/ but we do not really have anything changed in the `micropython` (git submodule) dir.

In fact, it can even be reproduced in cPython!

```python
lap1:$ python
Python 3.11.3 (main, May 10 2023, 12:13:23) [GCC 12.2.1 20220924] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> while True:
...     try:
...         for i in range(1000):
...             pass
...         raise(TypeError("Any error you actually want to catch"))
...     except:
...         pass
...
^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt
>>>
KeyboardInterrupt
>>>
KeyboardInterrupt
>>>
KeyboardInterrupt
>>>
```

Is this something we want to address? On cPython, it is possible to kill python externally, but in MicroPython, is it desired to have any higher priority in MicroPython nlr_raise() calls?

By adding [logging into `nlr_raise()`](https://github.com/micropython/micropython/blob/master/py/nlr.h#L177) we could observe that both Ctrl-C and the python code's error call `nlr_raise()`.

There is also not a lot of opportunity to inject a check for scheduled keyboard exception [`[1]`][1] [`[2]`][2] in nlr_raise():
https://github.com/micropython/micropython/blob/master/py/runtime.c#L1635

[1]: https://github.com/micropython/micropython/blob/master/py/scheduler.c#L50
[2]: https://github.com/micropython/micropython/blob/master/py/scheduler.c#L34

I am not entirely sure what is going on here, as once the `KeyboardInterrupt` is scheduled, it should be caught upon the next check while running the busy `try:` loop's code (the `for i in range()`)?

Congrats on being bug-to-bug compatible with cPython with a different implementation!

robert-hh, 2023-06-01:
To cite and old joke: "This is not a bug, it is a feature".
KeyboardInterrupt is the exception which is raised when Ctrl-C is pushed. Your open except: in the test code absorbs that one as well. To catch ctrl-c again, you have to add another exept clause, like:

except KeyboardInterrupt:
    raise
except:
    pass

closed 2023-06-01 by josuah

robert-hh, 2023-06-01:
All kudos go to Damien, who created MicroPython.

thyrrestrup, 2023-06-10:
You could look into the mp_sched_vm_abort mechanism if you want the abort to be uncatchable. This is only useful if you implement a custom handler on the C side.